steve-community/steve #664: Adding "Unknown RF ID" tag results in Unique Constraint SQL exception

XeroX1988, 2021-09-22:
When clicking the Add button for an Unknown RD ID tag I'm getting a SQL exception that the RF ID tag could not be added due to an unique contraint on ID Tag column .

Clicking Forget and adding a new RF ID tag with the same name also results in an already exists message.

System information:
Docker image based on the latest master branch

goekay, 2021-09-23:
You are trying to insert an RFID tag that exists in database. The "unknown Id tag" part of the interface lists incoming idtags that could not be found in database. It's just a helper. "Forgetting" an id tag there does not delete it from database.

You apparently have this id tag in database. Why are trying to insert it again?

XeroX1988, 2021-09-23:
Thanks for getting back to me Goekay. I've looked with the user that reported this, and the RF ID tag in question was not in the list of RF ID tags, so from a users perspective, it was not already in the system.

If needed I'm happy to retrace the steps taken by the user to get the system in the state that I found it in when reported.

goekay, 2021-09-23:
Interesting. If you could provide the logs from around that time, it would be helpful to understand what's going on.

goekay, 2021-11-07:
closing this issue due to its inactivity.

closed 2021-11-07 by goekay