steve-community/steve #666: OCPP charge point (TELLUS 4120) keeps dropping communication to steve

rishabhmehandru, 2021-09-25:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...3.4.5
Operating system  : ...ubuntu debian
JDK               : ...
Database          : ...mysql 5.7
```

#### Expected Behavior
Connection with websocket maintained all the time. 
...

#### Actual Behavior
Connection to machines gets dropped.When physically machine interface is rebooted, connection is re-estabilished
...

#### Steps to Reproduce the Problem

  1. ...More than one charge point connected
  2. ...One just drops within a few hours
  3. ...

#### Additional context

...

goekay, 2021-10-02:
how often do these drops occur? steve should not cause any connection drops. are you sure that steve's network is stable? are you sure that charging stations maintain a stable network connectivity? can you post any logs for more context?

rishabhmehandru, 2021-10-03:
Yes, I think we need machine to machine SIM

goekay, 2021-10-03:
> Yes, I think we need machine to machine SIM

i don't know how to start any investigation with this bit of information.

i assume that your problem is not because of steve, but other components. i cannot help you there.

rishabhmehandru, 2021-10-09:
Hi

I switched to machine2machine SIM but I am still seeing dropping off of one of the chargers. Will you be willing to look at the log and help me figure out why the charger is dropping?

Best
Rishabh

rishabhmehandru, 2021-10-09:
[steve.log](https://github.com/RWTH-i5-IDSG/steve/files/7316018/steve.log)

rishabhmehandru, 2021-10-10:
> here is my report:
>
>    - the logs cover the operation of approx. 16 hours and 30 mins
>    - during this period 423 messages were sent successfully
>    - 14 "abnormal connection close" cases
>    - the error codes are 1006 and 1011
>    - there is no pattern to why these connections drop. in one case, the
>    dropped connection was open for a couple of milliseconds (
>    sessionId=bee9343d-6ca2-1b7a-5733-14ab885151b0). in another case, the
>    connection was open for almost 6 hours (
>    sessionId=2c30dcca-dee2-f49d-148a-cc4130a00485) without issues and
>    then suddenly it was being dropped.
>    - most of the drops were during the night (from 00:00 to 08:00). after
>    8:00 there are only 3 cases. however, i dont know that to do with this info.
>
> the logs of steve do not tell anything further. there is no indication
> that steve is at fault. if you want to continue your investigation and find
> the root cause, you should look at the logs of the charging station.
> hopefully you can find the reasons there.
>
> my strong assumption is that the instability of wireless networks can
> cause connections being dropped and reopened from time to time. 14
> connection drops while 423 messages were sent successfully within 16,5
> hours can be considered acceptable. the missing patterns in connection drop
> behavior support my assumption.

Hi

Thank you so much for your help. I will investigate further on the machine
side to see why this could have happened.

Best
Rishabh

closed 2021-10-10 by goekay