steve-community/steve #669: Missing many @Override annotations, also name shadowing

evanbross, 2021-09-29:
#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

My group has very strict rules about java warnings & safety, based on past problems.
Missing @Override annotations and name shadowing are flagged as errors to prevent problems from creeping in during SDLC
...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

goekay, 2021-10-02:
can you give code examples for these issues? or better: please make a PR that addresses these issues. we would welcome that!

evanbross, 2021-10-03:
I'm always reluctant to add things back to an open source repo since my
java formatter may not agree with the existing format

goekay, 2021-10-03:
hey, i am making a general code refactoring round in [code_refactor](https://github.com/RWTH-i5-IDSG/steve/commits/code_refactor) branch. however, i am still unsure whether your points are being covered. your points are vague without concrete references to the code. can you please help me help you?

> I'm always reluctant to add things back to an open source repo since my
> java formatter may not agree with the existing format

why should just adding `@Override` or fixing name shadowing cause format issues? if there are issues because of formatting (which i am flexible about), i would comment in the PR. what is the big deal? i guess your group is using steve or building something on top of steve... why not being a little bit cooperative here in this OSS world? which brings me to my next point about the following sentence:

> My group has very strict rules about java warnings & safety, based on past problems.

considering my arguments above and your general reluctance of guiding or helping me, why should _this_ project even care about _your_ group's rules then? i, as the only developer of this project, did and do not have any problems.

evanbross, 2021-10-04:
I've disabled my auto-code formatter and have pulled a clean clone of the
project.
I added a few of the missing @Override annotations.
Is it ok to push to master or should I be creating a branch. I've observed
that some projects like to do it one way, some the other.

goekay, 2021-10-04:
> Is it ok to push to master or should I be creating a branch.

you will not be able to push to steve's master due to missing permissions.

"how to"s:
- https://gist.github.com/Chaser324/ce0505fbed06b947d962
- https://guides.github.com/introduction/flow/

goekay, 2021-11-07:
closing this issue due to its inactivity.

closed 2021-11-07 by goekay