steve-community/steve #682: the jar file starts and exits with code 0

denish-Goklani, 2021-10-18:
I ran the code as `  java -jar target/steve.jar  `

Log File says: 

```

[INFO ] 2021-10-18 12:56:35,953 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2021-10-18 12:56:36,069 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'DEV' profile

```

and that is all that it shows, post that my terminal shows 
`` Process finished with exit code 0  ``

Any help at earliest would be appreciated 😅

denish-Goklani, 2021-10-18:
A bit of debugging and it started working

dimitrido, 2021-12-19:
@denish-Goklani I am experiencing a similar behavior, what did you do to fix it?