steve-community/steve #697: connector_pk numbers not increasing by one

rdc-Green, 2021-11-08:
Hi,

I have a few charge points added to my system. When I add a new one, instead of the connector_pk increasing by an increment of 1, it increases by a random number (it seems) by about 200,000.

This doesn't appear to cause any issues yet, but I was wondering if this was a known issue (if it is even an issue) or if there is some way to fix it so that each time a connector is added it just increments by 1.

Thank you
Richard

goekay, 2021-11-08:
are you sure about this? it's hard to say from outside why it is like this, but it should not be like that.

rdc-Green, 2021-11-09:
i'm sure @goekay please see screenshot below of some connector_pks in cronological order:

[image]

Is there any part of the code you could suggest I take a look at to see why it is doing this?

goekay, 2021-11-09:
this field is managed by database since it is a primary key with auto increment. we can see from the screenshot that the PKs do not increase by 200k. 

the only explanation i can give: there are gaps between the values because you probably deleted some charging stations that had connectors that were deleted as well.

rdc-Green, 2021-11-09:
Hi,

i was just approximating with the 200k. As you can see it goes from 609 to 3483 and then 3932 to 347093. I haven't deleted that many charging stations.

I will keep looking to see if i can find a reason. Thank you for takinig the time to get back to me.

goekay, 2021-11-09:
> I haven't deleted that many charging stations.

maybe one station was misbehaving and therefore reported a lot of connectors to steve?

rdc-Green, 2021-11-09:
i suppose there is no way of finding the misbeahving station!

goekay, 2021-11-09:
you can look at the number of connectors per station:

```
select charge_box_id, count(*) as number_of_connectors
from connector
group by charge_box_id
order by number_of_connectors;
```

rdc-Green, 2021-11-09:
I tried that, but i have between 2-3 depending if its dual socket or not.

goekay, 2022-01-23:
i am closing this issue because there was no follow-up, and i doubt it is a fault of steve. @rdc-Green you can re-open it, if you have further info and evidence.

closed 2022-01-23 by goekay