steve-community/steve #702: Setting up ChargingProfile on RemoteStart

scobin3, 2021-11-17:
#### Checklist

- [x ] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
Hi SteVe team,

I'm wondering if there is a plan to add "ChargingProfile" field on the "RemoteStartTransaction.req" message?

Thank you,

...

gvtelio, 2022-12-14:
I am wondering the same. I am assuming if the chargingProfile is set on the RemoteStartTransaction request, it would be in effect, if the charging is started?

goekay, 2025-09-14:
hey everyone, this was done couple of days ago in https://github.com/steve-community/steve/pull/1809

closed 2025-09-14 by goekay