steve-community/steve #708: log4j version vulnerable to CVE-2021-44228

anthonyjmartinez, 2021-12-10:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : latest
Operating system  : any
JDK               : any
Database          : any
```

#### Expected Behavior

SteVe operates without dependency on libraries with published high-severity CVE

#### Actual Behavior

SteVe operates with dependency on libraries with published high-severity CVE
See https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-44228

and also https://logging.apache.org/log4j/2.x/security.html
...

#### Steps to Reproduce the Problem
N/A

#### Additional context

Recommend to a) disable JNDI entirely in the log4j config, and b) bump the version for the dependency on log4j to 2.15.0+

...

goekay, 2021-12-10:
thanks for the heads up @anthonyjmartinez.

do i understand it correctly, that with the new dependency version disabling property is not needed because it becomes the default?

anthonyjmartinez, 2021-12-10:
@goekay - You're welcome. As for the removal of JNDI, I can't say for sure. JNDI is exploited frequently and I believe that if it is not strictly needed it should be explicitly disabled to protect against possible regressions later down the line. Having been bitten by mistakenly believing one thing was a default when it wasn't more than a few times I always favor being explicit when possible.

goekay, 2021-12-10:
agreed in principle. 

however, when i dug deeper i found that this flag `log4j2.formatMsgNoLookups` has actually been deprecated and is no longer used. therefore, there is no need to _explicitly disable it_.

sources:
- from https://logging.apache.org/log4j/2.x/changes-report.html#a2.15.0:
> Pattern layout no longer enables lookups within message text by default for cleaner API boundaries and reduced formatting overhead. The old 'log4j2.formatMsgNoLookups' which enabled this behavior has been removed as well as the 'nolookups' message pattern converter option. The old behavior can be enabled on a per-pattern basis using '%m{lookups}'.
- from the actual [PR ](https://github.com/apache/logging-log4j2/pull/607) and [actual code](https://github.com/carterkozak/logging-log4j2/blob/2731a64d1f3e70001f6be61ba5f9b6eb55f88822/log4j-core/src/main/java/org/apache/logging/log4j/core/util/Constants.java#L57-L68)
```
    /**
     * LOG4J2-3198 property which used to globally opt out of lookups in pattern layout message text, however
     * this is the default and this property is no longer read.
     *
     * Deprecated in 2.15.
     *
     * @since 2.10
     * @deprecated no longer used, lookups are only used when {@code %m{lookups}} is specified
     */
    @Deprecated
    public static final boolean FORMAT_MESSAGES_PATTERN_DISABLE_LOOKUPS = PropertiesUtil.getProperties().getBooleanProperty(
            "log4j2.formatMsgNoLookups", true);
```

goekay, 2021-12-11:
the changes are merged and i made a new release because of the importance of this.

thanks @anthonyjmartinez !

closed 2021-12-11 by goekay

anthonyjmartinez, 2021-12-14:
@goekay - this appears to be the gift that keeps on giving

https://logging.apache.org/log4j/2.x/security.html has notice of a new CVE related to the previous and mitigated by using 2.16.0.

Do you want me to create another issue?

goekay, 2021-12-15:
@anthonyjmartinez thanks for following this topic :) 

apparently, @benvia has a PR already, thanks!