steve-community/steve #710: After Update to 3.4.6 still 3.4.5 is shown in webinterface

RedPeace, 2021-12-11:
Hi,

#### Checklist

- [x ] I checked other issues already, but found no answer/solution
- [x ] I checked the documentation and wiki, but found no answer/solution
- [x ] I am running the latest version and the issue still occurs
- [ x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.5 / 3.4.6
Operating system  : Ubuntu
JDK               : ...
Database          : MariaDB
```

#### Expected Behavior

#### Actual Behavior

i've downloaded and extracted 3.4.6.
Then replaced the main.properties

While compiling 3.4.5 is shown and the webinterface show 3.4.5 too.

Is it a bug or my fault?

#### Steps to Reproduce the Problem

  1. Download 3.4.6
  2. Extract
  3. Replace main.properties
  4. compile
  5. start

Regards!
Gero

goekay, 2021-12-11:
it is my fault because i forgot to update the version string. fix incoming. however, it does not mean that the release is faulty.

goekay, 2021-12-11:
thanks @RedPeace. the release is updated.

closed 2021-12-11 by goekay