steve-community/steve #713: When a client tries to charge, the connection to the database is lost.

jvmartinez-coves, 2021-12-15:
#### Checklist

- [:heavy_check_mark: ] I checked other issues already, but found no answer/solution
- [ :heavy_check_mark:] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.2
Operating system  : Ubuntu 20.10
JDK               : 11
Database          : Mysql 8.0
```

#### Describe the problem

Hello, good afternoon. Thank you for your work with Steve.
I have the following problem:
I have Steve 3.4.2 and a Mysql 8.0 database hosted on AWS Lightsail (1GB RAM, 1 vCPU, 40GB SSD (38.16GB used)). And I have the following issue for about a month: When a customer goes to charge the car, sometimes the error shown in the steveLog.txt file appears and the database is restarted, as shown in the awsLog.png file. The connection is lost and the database is restarted.

I am lost and don't know what to do.

Thank you very much for your time.
[steveLog.txt](https://github.com/RWTH-i5-IDSG/steve/files/7719339/steveLog.txt)
[image]

goekay, 2021-12-19:
there seem to be a general connectivity issue between steve and the database. it's hard to tell why.

maybe @csamsel can chime in.

goekay, 2022-05-26:
closing due to inactivity.

closed 2022-05-26 by goekay