steve-community/steve #727: Log file for metervalues

sea-blorin, 2022-01-14:
Is it possible to generate a Log csv file from the table of MeterValues for all transactions?

goekay, 2022-01-14:
hey, thanks for the request. yes, it is possible. however, i want to understand your use case. do you want a batch operation that generates this file for all metervalues of all transactions in "one go" (your question reads like it)? or do you want to be able to export metervalues on the transaction details page?

sea-blorin, 2022-01-17:
Thanks for your reply, I want export all transactions with details, for a research work of the state of charge and discharge of a V2G veicle. I want to export all transactions to a CSV files.

sea-blorin, 2022-01-17:
I want, if possible, to be able to export metervalues on the transaction details page, or be able to access to a csv files located in the server DB