steve-community/steve #736: HIKARI maxLifetime vs MariaDB wait_timeout

MysticJay, 2022-01-15:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.5 Docker. (I have questions about the upgrade procedure and Log4J is not important here)
Operating system  : QNAP QTS 4.5.4 with 
JDK               : ? as in the Docker created form docker-compose.yml
Database          : MariaDB 10 from the generated Docker
```
in the logs I find this:
```
{"log":"[WARN ] 2022-01-15 09:01:09,333 com.zaxxer.hikari.pool.PoolBase (qtp247162961-374) - HikariPool-1 - Failed to validate connection com.mysql.cj.jdbc.ConnectionImpl@2826e648 (No operations allowed after connection closed.). Possibly consider using a shorter maxLifetime value.\n","stream":"stdout","time":"2022-01-15T09:01:09.334185851Z"}
```
The advise found in Google is that the connection is terminated from MariaDB within 600 sec, while the value maxLifetime is bigger than those 600s. maxLifetime (in ms) should be set to 590000 to have a smaller value and force SteVe to reconnect before trying again.
(https://stackoverflow.com/questions/46830159/hikari-failed-to-validate-connection-because-connection-is-closed)

Where should this value be configured?

MysticJay, 2022-01-16:
Proposal for BeanConfiguration.java:
after:
```    private void initDataSource() {
        SteveConfiguration.DB dbConfig = CONFIG.getDb();

        HikariConfig hc = new HikariConfig();

        // set standard params
        hc.setJdbcUrl("jdbc:mysql://" + dbConfig.getIp() + ":" + dbConfig.getPort() + "/" + dbConfig.getSchema());
        hc.setUsername(dbConfig.getUserName());
        hc.setPassword(dbConfig.getPassword());
```
add
`hc.setMaxLifetime (580000);`
hope that's it (sorry, I am not familiar with anything in this project, so for the time being I will not create a PR for this one-line fix)

M.

skullmonkey911, 2022-03-03:
Hi,

I have the same issue as MysticJay describes.

I've two questions about it: 

1. @MysticJay : Is it working for you with this config?
2. @goekay : Is this workarround sound in your point of view?

Kind regards,

Stephan

goekay, 2022-03-03:
hey all, thanks for bringing this up. i understand the problem and am okay with the suggested solution. i will make the changes shortly.

goekay, 2022-03-07:
referenced commit addresses the issue, and i am therefore closing it

closed 2022-03-07 by goekay