steve-community/steve #737: Setup/update Docker

MysticJay, 2022-01-15:
Hi,
I started with SteVe 3.4.5 using the Docker-setup.
docker-compose is creating a weird naming and now that I want to update to 3.4.8 this makes no sense any more...

The TGZ was extracted to a folder named steve-steve-3.4.5
The APP created is named steve-steve-345 with two containers
- steve-steve-345_app_1
- steve-steve-345_db_1
everything running fine

IMHO the Readme is missing 
```mv steve-steve-345 steve```
to rename the folder and create The APP "steve" with containers "steve_app_1" and "steve_db_1"

Now I am stuck with the Update to 3.4.8.
should I rather follow my own thoughts and rename the extracted TGZ to steve, then export an import the DB to the new APP "steve"
OR
overwite all files in steve-steve-3.4.5 befor running compose again
OR
am I totally off the track? (then please give me some advise)

BTW: is there any Forum/Community/Chat/Telegram-Group)?

Thanks
M.

MysticJay, 2022-01-15:
docker-compose.yml reads
```app:
   build: .
```
shouldn't this rather be 
```app:
   build: steve:3.4.5
```
IMHO this would also improve the docker compose and would be independent from the name of the source folder.

goekay, 2024-08-07:
closing due to inactivity.

closed 2024-08-07 by goekay