steve-community/steve #746: Evbox update firmware command

testocpp, 2022-01-23:
Hi all, 

Have been using Steve for research purpose on on-off basis and its a very good system. I have been trying to update firmware on an Evbox ELVI charger for some time and somehow its not accepting the command. Does any one happen to know what command structure happens to look like? 

I was able to get diagnostic file command which following ftp://username:password@Ip_address/folder structure but have been struggling a lot with firmware command. 

Thanks for all the help in advance!

Cheers

lategoodbye, 2022-01-23:
@testocpp What makes you think this issue is caused by Steve? Can you provide the OCPP log (remove the credentials before)?

goekay, 2022-01-23:
> ... and somehow its not accepting the command.

the message/command that steve sends is such a simple one: download the firmware from URL X at date/time Y and you can retry Z times.

i cannot imagine that something is wrong with the message structure itself. however, the values you pick for X, Y and Z are important, and some charging station implementations might be picky and/or uncooperative. probably the most important parameter is the URL and what the station recognizes as "acceptable" is specific to that station. i cannot help you debug the situation unfortunately. it might help to look at station logs and see how it responds to the command.

jpaquete, 2022-02-14:
> Hi all,
>
> Have been using Steve for research purpose on on-off basis and its a very good system. I have been trying to update firmware on an Evbox ELVI charger for some time and somehow its not [...]

Hello testocpp, I am also having trouble performing the GetDiagnostics on the EVBOX. What format did you use for the URI ?. I have a working ftp server but the file never gets there. Any help I woulb be gratefull

testocpp, 2022-02-15:
> 

Hi mate, i didnt knew any better place to post this question. Charger does accept the command but nothing happens.

testocpp, 2022-02-15:
> Hello testocpp, I am also having trouble performing the GetDiagnostics on the EVBOX. What format did you use for the URI ?. I have a working ftp server but the file never gets there. Any help I [...]

Hello, please use ftp://[username]:[password]@[IPaddress of FTP server]/[folder]/

jpaquete, 2022-02-15:
> Hello, please use ***@***.*** of FTP

Hey many tks for response!. Tryed that already several times with url/ip
etc. I have a perfectly woking FTP server on windows. What is your
environment ? Running out of options here :(. Is it possible that the EVBOX
charger has ftp communication blocked ?
Thanks!

testocpp, 2022-02-15:
Happy to test diagnotic on my ftp server. Can you PM me and we can work out a time to do it together.

jpaquete, 2022-02-15:
Hey, thanks very much! I will be glad to help you on any issue if I can. I
dont see where to send PM here on github. But I have updated my profile to
show contact email.

closed 2023-08-20 by goekay