steve-community/steve #749: Invalid SNI Request

andle, 2022-01-26:
#### Checklist

- [ x] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : Ubuntu 20
JDK               : 11.0.13
Database          : MariaDB 10.3.32
```

#### Expected Behavior

Login page to Web Interface accessible, and chargers connect to server.

#### Actual Behavior

SNI Errors:

'WARN ] 2022-01-26 17:33:59,802 org.eclipse.jetty.server.HttpChannel - handleException /steve/manager/home org.eclipse.jetty.http.BadMessageException: 400: Invalid SNI
[WARN ] 2022-01-26 17:34:00,301 org.eclipse.jetty.server.HttpChannel - handleException /favicon.ico org.eclipse.jetty.http.BadMessageException: 400: Invalid SNI'

#### Steps to Reproduce the Problem

  1. Pulled latest from git
  2. Clean, build, restart.
  3. Go to web url...

#### Additional context

Logs show SNI errors for both the web connection as well as charger socket connections.
Tried updating certificate, rebuilding, still getting error.

andle, 2022-01-26:
Went back to 6c4c637fc4807bd19a8d4576ca77d839aacb7086 from Dec 25th, and problem is not present.

goekay, 2022-01-26:
thanks @andle, i can reproduce.

the problem happens with HEAD after we migrated from 9.x to jetty 10.x. 

the problem is no more there, if i checkout a commit before the respective changes in https://github.com/RWTH-i5-IDSG/steve/pull/747

i am looking into it.

goekay, 2022-01-27:
if we modify the following line

https://github.com/RWTH-i5-IDSG/steve/blob/5619987870c17c3c8c88d6e7ff2faf406272dcde/src/main/java/de/rwth/idsg/steve/JettyServer.java#L144

to 

```
httpsConfig.addCustomizer(new SecureRequestCustomizer(false)); 
```

it will set `sniHostCheck` to false instead of true, basically disabling SNI. i am not sure about the security implications of this, since i am not that familiar with SNI and SSL mechanisms.

goekay, 2022-01-29:
hey @andle, after further inspection, i found out that the SNI handling was reworked in jetty 10.x for tighter checking: https://github.com/eclipse/jetty.project/issues/5379

what was 200 OK back then, should have been 400. therefore, i consider it "works as intended". this means that you probably should recreate your certificates with the expected requirements.

closed 2022-02-08 by andle