steve-community/steve #759: Can't open home page from other device in the same local network

DilyanIvanov, 2022-02-13:
Hello everyone,
I just installed SteVe and can't open home page from other device in the same local network. If I type http://localhost:8080 on host server everything works just fine, but if I type http://172.16.1.230:8080 (172.16.1.230 is the IP of the machine where SteVe is installed), from other machine in the same local network, it is unable to connect. I'm with Ubuntu 20.04, installed and conf all prerequisites for Ubuntu and configure UFW
https://github.com/RWTH-i5-IDSG/steve/wiki/Prepare-Ubuntu-VM-for-SteVe
Part of my conf:

# Jetty configuration
#
server.host = localhost
server.gzip.enabled = true

# Jetty HTTP configuration
#
http.enabled = true
http.port = 8080

My EVSE Controller log
[E][WiFiClient.cpp:132] connect(): socket error on fd 64, errno: 104, "Connection reset by peer"
[ClientSocket] Disconnected!

What could be the problem?

goekay, 2022-02-13:
hey, instead of

> server.host = localhost

can you please setting it to

> server.host = 172.16.1.230

DilyanIvanov, 2022-02-13:
I changed server.host = localhost to server.host = 172.16.1.230 then restart SteVe, but the result is the same. I noticed that after the server.host change ocpp endpoints are again with 127.0.0.1 instead 172.16.1.230
Is that ok?

Access the web interface using
- http://127.0.0.1:8080/steve/manager/home
SOAP endpoint for OCPP
- http://127.0.0.1:8080/steve/services/CentralSystemService
WebSocket/JSON endpoint for OCPP
- ws://127.0.0.1:8080/steve/websocket/CentralSystemService/(chargeBoxId)

goekay, 2022-02-13:
> ... then restart SteVe ...

where is the config file? is it an external config file? if it is within the project directory, you need to rebuild the project for the changes to take effect.

DilyanIvanov, 2022-02-13:
Now it works! The key was to rebuild the project. Thank you very much for your help @goekay

closed 2022-02-13 by DilyanIvanov