steve-community/steve #769: Formation Violation 29.02.2022

Rouffy, 2022-03-01:
SteVe Version     : 3.4.8
Operating system  : Ubuntu 20.04.
JDK               : OpenJDK 11.0.13
Database          : MariaDB 10.3.32

Hi,
i'm getting the following error today and i dont know if its SteVe or the Box which is an ABB Terra AC
Systemtime in Ubuntu is set to UTC, same goes for MariaDB.
[INFO ] 2022-03-01 07:02:31,299 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=TACW22xxxxxxxxxx, sessionId=85ec53a4-47bb-2535-c423-422097183dbc] Sending: [4,"8527040","FormationViolation","The payload for action could not be deserialized",{"errorMsg":"Cannot parse \"2022-02-29T07:02:33.000Z\": Value 29 for dayOfMonth must be in the range [1,28] (throug..."}]

goekay, 2022-03-01:
hey, the important part is the error msg:

> Cannot parse "2022-02-29T07:02:33.000Z": Value 29 for dayOfMonth must be in the range [1,28] 

the station apparently sent a message using 2022-02-29 which is not valid. february in 2022 does not have 29 days.

Rouffy, 2022-03-01:
Hi,

thanks, i saw that the date is wrong, i didnt know if its SteVes fault or the station, so i have to look at the station, thanks :-)

csamsel, 2022-03-01:
always funny to see what kind of problems charging stations have...

Rouffy, 2022-03-01:
Firmware Update on all ABB Stations did the trick :-D
Thanks

closed 2022-03-01 by Rouffy