steve-community/steve #776: No mapping for GET /steve/websocket/CentralSystemService

carminegreen, 2022-03-09:
After starting the server with docker and nginx, I try to connect to the websocket link present in the readme and I get this error

[image]

### **Server log**
[image]

### P.S.
The nginx configuration is well structured because I can navigate the management system, therefore this is not the problem.

goekay, 2022-03-10:
can you provide more context? what version of steve, or HEAD of the repo? other log lines?

csamsel, 2022-03-10:
you probably didnt follow the instruction how to use steve behind a reverse proxy: https://github.com/RWTH-i5-IDSG/steve/wiki/FAQ#how-can-i-run-steve-behind-a-reverse-proxy

carminegreen, 2022-03-11:
I'm currently using steve 3.4.8 through Docker. I'm using Nginx as the reverse proxy, and I have steve running on a sublocation of my main domain (like: {{my.domain.com}}/steve):

```
location /steve {
  proxy_pass http://steve_app_1:8180;
}

location /steve/websocket/CentralSystemService {
  proxy_pass http://steve_app_1:8180/steve/websocket/CentralSystemService;
  proxy_http_version 1.1;
  proxy_set_header Upgrade $http_upgrade;
  proxy_set_header Connection "upgrade";
  proxy_connect_timeout 1d;
  proxy_send_timeout 1d;
  proxy_read_timeout 1d;
}
```

If I try to connect using wscat, I get the aformentioned error (404, no GET mapping). If I add my chargebox id, I get a 500 response saying that the OCPP version is not specified in the chargebox (which should be set by the ws itself afaik).

Clearly I'm doing something wrong, what could it be?

goekay, 2022-03-11:
for a successful websocket handshake, you have to
- provide chargeboxid in the url, and
- this chargeboxid has to be registered (i.e. in database of steve)
- set the websocket subprotocol to the ocpp version.

the websocket protocol usage in ocpp is not just a "connect to this url" kind of a logic. it has additional ocpp-related constraints. please check the ocpp spec.

and please see the `return false` cases in https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/OcppWebSocketHandshakeHandler.java

carminegreen, 2022-03-14:
I entered the id in the database through the steve management software, then I made the call: /steve/websocket/CentralSystemService/{{id}}
This is the result: 
[image]

goekay, 2022-03-14:
because you still did not do what i wrote above:

> set the websocket subprotocol to the ocpp version.

carminegreen, 2022-03-14:
How set the websocket subprotocol to the ocpp version?

goekay, 2022-03-14:
sorry, but github issues are not a customer service mechanism. you have to refer to the documentation of your client.

setting a subprotocol within the upgrade request is part of the websocket standard. ocpp utilizes it for the clients the specify their ocpp implementation.