steve-community/steve #777: Installation/setup process is NOT repeatable

204maker, 2022-03-10:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
I am not expert in Linux.
I am trying to setup SteVe-3.4.8 on a Raspberry Pi 3 running in pre-installed armf Ubuntu 20.04 server.
I followed the Ubuntu setup to the compilation process and I successfully compiled the source.

However, it fails everytime I start to run SteVe server.
...

#### Describe the solution you'd like
Is there a more detailed instruction how to fix the smaller issues like:
-setting up wifi connection with DHCP disabled?
-setting up main.properties config?
-setting up UTC time where it is needed?
...

#### Describe alternative solutions or features you've considered 
An example SteVe server setup might be helpful.

For example, I would like to setup a SteVe server on my device (i.e. Raspberry) with the network details below:
RPi wlan0 IP: 192.168.0.15

How would I use this information to setup my RPI as a SteVe server? 
...

#### Additional context

...

shadsnz, 2022-03-10:
When you say "it fails everytime I start to run SteVe server", what exactly happens?  Do you see an error message?

204maker, 2022-03-10:
Here's a photo showing it failed to start SteVe server.
[image]

I don't have a photo of the log file but I will share it here when I get access to the Raspberry Pi again.

goekay, 2022-03-10:
hey, without taking a look at the log file it's hard to say what the cause is.

> setting up UTC time where it is needed?

maybe this could help: https://github.com/RWTH-i5-IDSG/steve/wiki/FAQ#how-to-fix-the-exception-mysql-and-java-are-not-using-the-same-time-zone

204maker, 2022-03-10:
Here is the log file:
[image]

goekay, 2022-03-10:
i don't know how to interpret this log file. there should be more log entries, especially from jetty. there is a big chunk missing which could have been helpful to diagnose.

for comparison, when i build the HEAD of the repo without any modifications, and start and stop steve i see the following:

```
[INFO ] 2022-03-10 17:55:42,576 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2022-03-10 17:55:42,670 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2022-03-10 17:55:42,685 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2022-03-10T16:55:42.675Z
[INFO ] 2022-03-10 17:55:42,921 org.eclipse.jetty.server.Server - jetty-10.0.7; built: 2021-10-06T19:34:02.766Z; git: da8a4553af9dd84080931fa0f8c678cd2d60f3d9; jvm 11.0.14+9
[INFO ] 2022-03-10 17:55:43,146 org.eclipse.jetty.server.handler.ContextHandler.steve - No Spring WebApplicationInitializer types detected on classpath
[INFO ] 2022-03-10 17:55:43,161 org.eclipse.jetty.server.session.DefaultSessionIdManager - Session workerName=node0
[INFO ] 2022-03-10 17:55:43,172 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring root WebApplicationContext
[INFO ] 2022-03-10 17:55:43,173 org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization started
[INFO ] 2022-03-10 17:55:43,774 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
[INFO ] 2022-03-10 17:55:43,931 com.zaxxer.hikari.pool.HikariPool - HikariPool-1 - Added connection com.mysql.cj.jdbc.ConnectionImpl@1cb9ef52
[INFO ] 2022-03-10 17:55:43,933 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
[INFO ] 2022-03-10 17:55:45,898 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2022-03-10 17:55:46,309 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP12
[INFO ] 2022-03-10 17:55:46,321 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService from class ocpp.cs._2012._06.CentralSystemService
[INFO ] 2022-03-10 17:55:46,370 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP15
[INFO ] 2022-03-10 17:55:46,371 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService from class ocpp.cs._2015._10.CentralSystemService
[INFO ] 2022-03-10 17:55:46,410 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP16
[INFO ] 2022-03-10 17:55:46,411 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2022-03-10 17:55:46,417 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemService
[WARN ] 2022-03-10 17:55:48,134 org.springframework.security.config.annotation.web.builders.WebSecurity - You are asking Spring Security to ignore Ant [pattern='/static/**']. This is not recommended -- please use permitAll via HttpSecurity#authorizeHttpRequests instead.
[INFO ] 2022-03-10 17:55:48,136 org.springframework.security.web.DefaultSecurityFilterChain - Will secure Ant [pattern='/static/**'] with []
[WARN ] 2022-03-10 17:55:48,137 org.springframework.security.config.annotation.web.builders.WebSecurity - You are asking Spring Security to ignore Ant [pattern='/views/**']. This is not recommended -- please use permitAll via HttpSecurity#authorizeHttpRequests instead.
[INFO ] 2022-03-10 17:55:48,137 org.springframework.security.web.DefaultSecurityFilterChain - Will secure Ant [pattern='/views/**'] with []
[INFO ] 2022-03-10 17:55:48,166 org.springframework.security.web.DefaultSecurityFilterChain - Will not secure any request
[INFO ] 2022-03-10 17:55:48,336 org.hibernate.validator.internal.util.Version - HV000001: Hibernate Validator 6.1.7.Final
[INFO ] 2022-03-10 17:55:48,673 org.springframework.web.context.ContextLoader - Root WebApplicationContext initialized in 5499 ms
[INFO ] 2022-03-10 17:55:48,729 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring DispatcherServlet 'spring-dispatcher'
[INFO ] 2022-03-10 17:55:48,729 org.springframework.web.servlet.DispatcherServlet - Initializing Servlet 'spring-dispatcher'
[INFO ] 2022-03-10 17:55:48,730 org.springframework.web.servlet.DispatcherServlet - Completed initialization in 1 ms
[INFO ] 2022-03-10 17:55:48,746 org.eclipse.jetty.server.handler.ContextHandler - Started o.e.j.w.WebAppContext@5aa6202e{/steve,jar:file:steve/target/steve.jar!/webapp,AVAILABLE}
[INFO ] 2022-03-10 17:55:48,755 org.eclipse.jetty.server.AbstractConnector - Started ServerConnector@342287d1{HTTP/1.1, (http/1.1)}{127.0.0.1:8080}
[INFO ] 2022-03-10 17:55:48,757 org.eclipse.jetty.server.Server - Started Server@1827a871{STARTING}[10.0.7,sto=5000] @11943ms
[INFO ] 2022-03-10 19:14:33,204 org.eclipse.jetty.server.Server - Stopped Server@1827a871{STOPPING}[10.0.7,sto=5000]
[... cut ...]
[INFO ] 2022-03-10 19:14:33,280 org.eclipse.jetty.server.AbstractConnector - Stopped ServerConnector@342287d1{HTTP/1.1, (http/1.1)}{127.0.0.1:8080}
[INFO ] 2022-03-10 19:14:33,295 org.eclipse.jetty.server.handler.ContextHandler.steve - Destroying Spring FrameworkServlet 'spring-dispatcher'
[INFO ] 2022-03-10 19:14:33,333 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown initiated...
[INFO ] 2022-03-10 19:14:33,342 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown completed.

```

goekay, 2022-04-11:
there is no follow-up in this conversation. i therefore consider the issue resolved. i am closing it.

closed 2022-04-11 by goekay