steve-community/steve #778: How to run SteVe as a wireless server on a local network using Ubuntu 20.04?

204maker, 2022-03-10:


204maker, 2022-03-10:
**What needs to change on the _main.properties_ file by keeping all database and web interface credentials DEFAULT?**

# Just to be backwards compatible with previous versions, this is set to "steve",
# since there might be already configured chargepoints expecting the older path.
# Otherwise, might as well be changed to something else or be left empty.
#
context.path = steve

# Database configuration
#
db.ip = localhost
db.port = 3306
db.schema = stevedb
db.user = steve
db.password = changeme

# Credentials for Web interface access
#
auth.user = admin
auth.password = 1234

# Jetty configuration
#
server.host = 127.0.0.1
server.gzip.enabled = true

# Jetty HTTP configuration
#
http.enabled = true
http.port = 8080

# Jetty HTTPS configuration
#
https.enabled = false
https.port = 8443
keystore.path =
keystore.password =

# When the WebSocket/Json charge point opens more than one WebSocket connection,
# we need a mechanism/strategy to select one of them for outgoing requests.
# For allowed values see de.rwth.idsg.steve.ocpp.ws.custom.WsSessionSelectStrategyEnum.
#
ws.session.select.strategy = ALWAYS_LAST

# if BootNotification messages arrive (SOAP) or WebSocket connection attempts are made (JSON) from unknown charging
# stations, we reject these charging stations, because stations with these chargeBoxIds were NOT inserted into database
# beforehand. by setting this property to true, this behaviour can be modified to automatically insert unknown
# stations into database and accept their requests.
#
# CAUTION: setting this property to true is very dangerous, because we will accept EVERY BootNotification or WebSocket
# connection attempt from ANY sender as long as the sender knows the URL and sends a valid message.
#
auto.register.unknown.stations = false

### DO NOT MODIFY ###
steve.version = ${project.version}
git.describe = ${git.commit.id.describe}
db.sql.logging = false
profile = prod

goekay, 2022-03-10:
did you try changing the IP in `server.host = 127.0.0.1`? or even better: https://github.com/RWTH-i5-IDSG/steve/wiki/Configuration#multiple-interfaces

goekay, 2022-05-26:
closing due to inactivity.

closed 2022-05-26 by goekay