steve-community/steve #779: 2.0.1 Support

alpercelerce, 2022-03-10:
Is there any plan for adding OCPP 2.0.1 support?

goekay, 2022-03-10:
pls see https://github.com/RWTH-i5-IDSG/steve/issues/513

wfe-keAl, 2022-03-11:
Hi there,

the implementation of OCPP 2 in the near future would be an advantage. Volkwagen offers charging points with OCPP 2.0J support. See attachment. All of the cars from volkswagen and their brands (Seat, Skoda, Cupra, Audi, ...) will support PnC soon, i think.

I would appreciate it if OCPP 2 comes to SteVe.

P.S. thanks for the great work. SteVe is awesome.
[image]

goekay, 2022-03-14:
hey @wfe-keAl thanks for your kind words.

i, obviously and for sure, see the value in an ocpp 2.0 implementation. however, when compared to, for example, version 1.6 the spec got longer and more comprehensive. it's overwhelming! i just cannot even find the time to grok the spec — let alone figure out how to integrate it into steve and then actually implement it.

csamsel, 2022-03-18:
> All of the cars from volkswagen and their brands (Seat, Skoda, Cupra, Audi, ...) will support PnC soon, i think.

Only newer ones, based on E³1.1/MEB (ID.3...) or E³1.2/PPE (Taycan). E³1.0/MQBevo (Golf) wont.

lategoodbye, 2022-04-01:
@wfe-keAl So your motivation of OCPP 2.0.1 is just PnC support?

The reason behind my question is the fact that PnC can be backported to OCPP 1.6J:
https://www.openchargealliance.org/uploads/files/ocpp_1_6_ISO_15118_v10.pdf

wfe-keAl, 2022-04-07:
@lategoodbye 

No. Not at all. PnC ist a quiet nice feature. But OCPP 2.0.1 got so much more to offer. For example: device management, security, smart charging functionality and display and messaging support.

But the main reason why i would appreciate it that Steve will support OCPP 2.01 is that some vendors (like VW with the I.Charger) only supports OCPP 2. OCPP 2 isnt backward compatible to 1.6 or is it? I dont finde any clue that it is or that the I.Charger support 1.6 as well. But i hopefully can check this in the near future. I will post my results.

greets keAl

lategoodbye, 2022-04-07:
@wfe-keAl 

Thanks for your explanation. OCPP 2.x is a complete redesign and not backward compatible. For example before OCPP 2.x the transaction id was generated by the OCPP backend and starting with OCPP 2.x it's generated by the charge point.

In my opinion some aspects of OCPP 2.x are far too complex, which prevents the charge point manufacturer from implementation.

goekay, 2022-04-07:
> OCPP 2.x is a complete redesign and not backward compatible. 

this is one of the reasons i am reluctant to commit to implementing it in steve.

so far, the versions were additive in nature. ocpp 1.6 includes 1.5 which includes 1.2. therefore, we implemented the business logic for the "most feature rich" version which is ocpp 1.6. the other versions are just mapping requests/responses upwards and downwards between versions.

as far as i understood ocpp 2.x is a departure from that.