steve-community/steve #790: Steve log file

snabo1988, 2022-03-28:
Hi, 
I have a problem with log file. Currently I use the newest version of Steve. I build a new container on my PC via docker. On my server log file is unavailable. I use the default configuration.

XeroX1988, 2022-03-31:
If you want the "Logs" page/button to work in a docker container approach, this is what I did to make it work:

- Run SteVe as a dedicated user (steve in my case)
- Make sure a home directory for the SteVe user is created/present
- Replace src/main/resources/config/docker/log4j2.xml with the following file

<?xml version="1.0" encoding="UTF-8"?>
<Configuration status="WARN">

    <Properties>
        <Property name="logPattern">[%-5p] %d %c (%t) - %m%n</Property>
    </Properties>

    <Appenders>
        <RollingRandomAccessFile name="FILE"
                                 fileName="${sys:user.home}/logs/steve.log"
                                 filePattern="${sys:user.home}/logs/steve-%d{yyyy-MM-dd}-%i.log.gz"
                                 immediateFlush="false"
                                 append="true">
            <PatternLayout pattern="${logPattern}"/>
            <Policies>
                <SizeBasedTriggeringPolicy size="10 MB"/>
                <TimeBasedTriggeringPolicy interval="1" modulate="true"/>
            </Policies>
            <DefaultRolloverStrategy max="20"/>
        </RollingRandomAccessFile>
    </Appenders>

    <Loggers>
        
        
        <AsyncLogger name="org.jooq.Constants" level="WARN"/>

        <AsyncLogger name="org.apache.cxf" level="INFO"/>
        <AsyncLogger name="org.eclipse.jetty" level="INFO"/>
        <AsyncLogger name="com.zaxxer.hikari" level="INFO"/>
        <AsyncLogger name="org.springframework" level="INFO"/>
        <AsyncLogger name="org.springframework.security" level="INFO"/>
        <AsyncLogger name="org.springframework.web.servlet" level="INFO"/>

        <AsyncLogger name="org.apache.jasper" level="WARN"/>
        <AsyncLogger name="org.apache.tomcat.util" level="WARN"/>
        <AsyncLogger name="jndi" level="WARN"/>

        <AsyncRoot level="INFO">
            <AppenderRef ref="FILE"/>
        </AsyncRoot>
    </Loggers>
</Configuration>

- Compile SteVe
- Start SteVe, log page should work for you now

goekay, 2022-05-03:
@XeroX1988 thanks. @snabo1988 is your issue resolved?

goekay, 2022-05-26:
closing due to inactivity.

closed 2022-05-26 by goekay

snabo1988, 2022-05-27:
Hi, sorry for my delay. Problem wasn't solved. All time logs file aren't available.