steve-community/steve #791: Spring4Shell Vulnerability

XeroX1988, 2022-03-31:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : latest
Operating system  : Ubuntu 20.04 LTS
JDK               : default-jre (openJDK 11)
Database          : MariaDB
```

#### Expected Behavior
Spring4Shell vulnerability is mitigated in SteVe

#### Actual Behavior
Current implementation of SteVe might be vulnerable to Sping4Shell vulnerability, details:
https://www.cyberkendra.com/2022/03/springshell-rce-0-day-vulnerability.html

goekay, 2022-03-31:
hey @XeroX1988 thanks. i am aware and following the updates.

just a couple of hours ago: https://spring.io/blog/2022/03/31/spring-framework-rce-early-announcement

i will update spring dependencies as a fix.

goekay, 2022-03-31:
more context: https://en.reddit.com/r/java/comments/tscqlg/spring4shell_0day_rce_on_spring_core/

goekay, 2022-03-31:
i made a new release: https://github.com/RWTH-i5-IDSG/steve/releases/tag/steve-3.4.9

XeroX1988, 2022-03-31:
Thanks Goekay, re-building my containers with the latest version as we speak.