steve-community/steve #796: Unable to Stop  Remote Transaction

subhankar007b, 2022-04-04:
Hi there,
I'm able to start Remote Transaction but after that, it's unable to store transaction data in the database. And I'm unable to stop transactions. Please if someone guides me on what I'm doing wrong will be helpful.

Thanks in Advance.

goekay, 2022-04-04:
> it's unable to store transaction data in the database.

why is this? can you post logs?

obviously, if the transaction cannot be stored in database, you will not be able to stop it since there will not be any transaction to reference.

subhankar007b, 2022-04-04:
[image]
after giving the command remotestarttransaction, charger gets started but after that, in the database no value gets stored. and im unable to stop that transaction

goekay, 2022-04-04:
assuming that the logs are complete, your charging station appears to not send a StartTransaction message after which steve would have stored the transaction in database.

this was a known problem for a manufacturer/model/firmware. is your charging station, by any chance, one of the mentioned ones here? https://github.com/RWTH-i5-IDSG/steve/issues/740

subhankar007b, 2022-04-04:
i have TELLUS POWER charger

goekay, 2022-04-04:
well, i cannot be more helpful than this, please contact the manufacturer. the station does not send the StartTransaction message.

subhankar007b, 2022-04-04:
okay... thank you for your response.

closed 2022-04-04 by goekay

parvesh-shadow, 2022-09-27:
@subhankar007b we are looking for support on OCPP had you successfully implemented it. We are looking for some support on it. Ready to pay for it.