steve-community/steve #797: Readme.md with AWS lightsail for Nginx

rahulbhat13, 2022-04-07:
#### Describe the problem you are trying to solve

The proxy setup with Nginx server for websocket getting the issue ERRconnection set issue .

Would suggest some to guide and help on the documentation part .

#### Describe the solution you'd like

Have configured the Nginx in the lightsail instance 
And the configuration in /etc/nginx/Nginx.conf

server {

location /steve/ {
    proxy_pass http://static IP of the light sail instance /steve/;
}

location /steve/websocket/CentralSystemService {
    proxy_pass http://static IP of the light sail instance /steve/websocket/CentralSystemService;
    proxy_http_version 1.1;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection "upgrade";
    proxy_connect_timeout 1d;
    proxy_send_timeout 1d;
    proxy_read_timeout 1d;
}
}
...

#### Describe alternative solutions or features you've considered 

Would need some solution so I can configure the right proxy and test the websocket issue with postman since I am getting connection issue , I feel I am doing something wrong in the configuration and have to setup the port 8080 somewhere 

#### Additional context

...

rahulbhat13, 2022-04-09:
Can someone help with the issue with the web sockets since I have setup in the main.properties and could see the ws:// connection been setup in the aws Lightsail too , but when I try to use the postman or ws client I am getting it as connection closed. In the webapp also I have setup the charger accordingly using Steve 
[image]
But not able to ping them using the ws

goekay, 2022-04-10:
https://github.com/RWTH-i5-IDSG/steve/issues/776#issuecomment-1065045988

you cannot just connect via websocket to steve. there are some ocpp specific requirements.

you should be able to see the particular problem why websocket handshake did not happen in the logs.

rahulbhat13, 2022-04-10:
org.eclipse.jetty.util.thread.QueuedThreadPool$[Runner.run](http://runner.run/)([QueuedThreadPool.java:1038](http://queuedthreadpool.java:1038/)) ~[jetty-util-10.0.7.jar:10.0.7]
at [java.lang.Thread.run](http://java.lang.thread.run/)([Thread.java:829](http://thread.java:829/)) ~[?:?]
[ERROR] 2022-04-10 08:27:53,431 de.rwth.idsg.steve.ocpp.ws.OcppWebSocketHandshakeHandler - No protocol (OCPP version) is specified.
[ERROR] 2022-04-10 08:28:09,527 de.rwth.idsg.steve.ocpp.ws.OcppWebSocketHandshakeHandler - No protocol (OCPP version) is specified.
[ERROR] 2022-04-10 08:28:16,706 de.rwth.idsg.steve.ocpp.ws.OcppWebSocketHandshakeHandler - No protocol (OCPP version) is specified.
[ERROR] 2022-04-10 08:29:09,128 de.rwth.idsg.steve.ocpp.ws.OcppWebSocketHandshakeHandler - No protocol (OCPP version) is specified.

goekay, 2022-04-10:
yeah. ocpp version is missing in header. see my comment and ocpp spec for further inspection.

rahulbhat13, 2022-04-10:
@goekay https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/ocpp16/Ocpp16WebSocketEndpoint.java will this work ?

goekay, 2022-04-10:
there is no need to change something in steve. steve implements ocpp spec as specified.

you are not understanding the problem.

the http request for ws upgrade/handshake should contain a specific header with the value of ocpp version the station implements.

please look at the spec for the details of ws specific requirements.

rahulbhat13, 2022-04-10:
async def main():
    server = await websockets.serve(
        on_connect,
        '0.0.0.0',
        9000,
        subprotocols=['ocpp2.0.1']
    )
    logging.info("WebSocket Server Started")
    await server.wait_closed()

if __name__ == '__main__':
    asyncio.run(main()) you need to pass the subprotocols =['version'] ? from the charger ?

goekay, 2022-04-10:
you know steve does not support ocpp2.0.1, right?

the supported versions are pretty much "in your face" in the readme.

goekay, 2022-04-10:
i will stop commenting in this issue. 

and i am closing it. github issues are not a customer support channel. i am pretty sure yours is not about a fault/bug in steve.

closed 2022-04-10 by goekay

rahulbhat13, 2022-04-10:
I got the version : OCPP1.2S
OCPP1.2J
OCPP1.5S
OCPP1.5J
OCPP1.6S
OCPP1.6J
 which are supported I know , I had already mentioned the reference . This is not a business requirement to test with simulator the ws is not working