steve-community/steve #798: java.lang.NullPointerException

emry78, 2022-04-07:
#### Checklist

- [ /] I checked other issues already, but found no answer/solution
- [/ ] I checked the documentation and wiki, but found no answer/solution
- [ 3.4.8] I am running the latest version and the issue still occurs
- [ /] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.8
Operating system  : Ubuntu 21.04
JDK               : 11.0.13
Database          : 10.5.13
```

#### Expected Behavior
Can display number of connected jason charge point.
...

#### Actual Behavior
when click number of connected jason charge point button from home. After click it display "java.lang.NullPointerException ".

...
**from Home:**
[image]

**page after clicking number of connected jason charge point button** 
[image]

**log file:**
[image]

#### Steps to Reproduce the Problem

  1. go to home
  2. click number of connected jason charge point button 
  3. error come at that page.

#### Additional context
no
...

lategoodbye, 2022-04-09:
@emry78 I'm using Version 3.4.9 and also to connected charge points, but i don't see the NullPointerException.
Do you see something special regarding to the charge points?
Can you provide the full stacktrace of the exception (especially the beginning is important)?

goekay, 2022-04-10:
yes. this screenshot is not helpful because the root cause of NPE is not there. it just depicts the lower layers of the exception.

goekay, 2022-04-26:
since there is no further activity/discussion, i am closing this issue.

closed 2022-04-26 by goekay