steve-community/steve #799: Time Different At Connection Status web page

emry78, 2022-04-07:
#### Checklist

- [/ ] I checked other issues already, but found no answer/solution
- [/ ] I checked the documentation and wiki, but found no answer/solution
- [3.4.8 ] I am running the latest version and the issue still occurs
- [/ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.8
Operating system  : Ubuntu 21.04
JDK               : 11.0.13
Database          : 10.5.13
```

#### Expected Behavior
expected time connection same with system
...

#### Actual Behavior
Time setting all set. Please advice if have any setting need

[image]

time at Connection Status:
[image]

#### Steps to Reproduce the Problem

  1. go to home page
  2. select connection status
  3. At connection status the time not same as we set all time system.

#### Additional context

no

goekay, 2022-04-10:
in this table steve shows the date/time of the status message that the charging station sent. there is no intermediate logic or interpretation happening. i would assume that the station does not send proper ISO 8601 date/times. historically we had many issues where the stations had faulty implementations wrt. time zones.

goekay, 2022-04-26:
since there is no further activity/discussion, i am closing this issue.

closed 2022-04-26 by goekay