steve-community/steve #803: Improve transaction view for SignedData

lategoodbye, 2022-04-14:
#### Checklist

- [X] I checked other issues already and found no answer
- [X] I checked the documentation and found no answer
- [X] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

As a user i like to see the content of a SampledValue with ValueFormat = SignedData in the transaction view without using hexdecoding tools.

#### Describe the solution you'd like

It would be nice to have the data decode directly in the webpage or maybe as a mouseover overlay.

#### Additional context

[OMCF](https://github.com/SAFE-eV/OCMF-Open-Charge-Metering-Format) is a standard to include signed meter values according to German Calibration law / Eichrecht. Since OCMF is also JSON the data must be encoded within OCPP StopTransaction. So it makes use of ValueFormat = SignedData, which is specified in OCPP 1.6 as  "Data is represented as a signed binary data block, encoded as hex data." Another issue is that hexencoding tends too very longish output, see screenshots.

Example:
``
[INFO ] 2022-04-14 08:50:06,475 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=twin_leipzig_a, sessionId=83986789-edb4-5b65-bfaf-16a96474b081] Received: [2,"5bd5ee3d3e725d7ddaf3af6ec9b91b9b9d0b","StopTransaction",{"idTag":"1ec11c60","meterStop":936,"reason":"EVDisconnected","timestamp":"2022-04-14T06:50:06.392Z","transactionData":[{"sampledValue":[{"context":"Transaction.End","format":"SignedData","unit":"kWh","value":"4f434d467c7b224656223a22312e30222c224749223a22636861726765506f696e7456656e646f7220636861726765506f696e744d6f64656c222c224753223a2231323334353637383930222c224756223a22312e302e35222c225047223a225431363336222c224d56223a2249736b7261222c224d4d223a22574d334d34222c224d53223a225734313134353936222c224d46223a22322e3033222c224953223a747275652c22494c223a2248454152534159222c224946223a5b22524649445f504c41494e222c224f4350505f4e4f4e45225d2c224954223a2249534f3134343433222c224944223a223165633131633630222c224354223a22455653454944222c224349223a2244452a494e542a45434831323334353637383930222c225244223a5b7b22544d223a22323032322d30342d31345430363a34393a35362c3030302b303030302049222c225458223a2242222c225256223a302e39332c225249223a22312d623a312e382e30222c225255223a226b5768222c225254223a224143222c225354223a2247227d2c7b22544d223a22323032322d30342d31345430363a35303a30352c3030302b303030302049222c225458223a2272222c225256223a302e39332c225249223a22312d623a312e382e30222c225255223a226b5768222c225254223a224143222c225354223a2247227d5d7d7c7b225344223a22333034363032323130303966393432353130303638336362383636366564333764636663353066306364383864346464333633613463376538316135303934643633393533663964393630323231303063623865613765633063636663303837353861303130343663386236343135376163343263353336643335373366656430396235666463396165343630323439227d"}],"timestamp":"2022-04-14T06:50:06.000Z"}],"transactionId":2595}]
``
 
[image]
[image]

lategoodbye, 2022-04-15:
After sleeping a night and rethinking there is an issue. The reason why the data is hexencoded it is possibly binary. So presenting it to the user would be impossible for all use cases. For OCMF it should be easy to detected because the data always starts with "4f434d467c".

goekay, 2022-04-15:
converting a hexcode into a text is easy and universal. the output text can still be gibberish because the content can be anything depending on the use case. at this point, it is the user's (i.e. reader's) responsibility to make sense of the output.

lategoodbye, 2022-04-19:
Thanks a lot. Unfortunately i learned that a lot of implementors interpret the format = "SignedData" different from the OCPP 1.6 spec.

Interpretation from OCPP 1.6J:
meter value is hexencoded

Interpretation from Eichrecht implementors like Keba (see #212):
meter value is digitally signed and encoding type depend on implementor (quoted, hex or base64).

So i'm afraid that this feature could cause a regression with these implementations :-(

goekay, 2022-04-19:
alright. i will leave the branch as is, since it might still fit your use case. you can use it if you wish.

however, i want to cherry pick 333b8be16d338f7bcf6851d793d05a6d33b4a5bd into master since it is a general stylistic improvement IMO.

lategoodbye, 2022-04-20:
Yes, cherry picking the format fix is fine. Thanks

closed 2022-04-25 by goekay