steve-community/steve #813: Question: remote db access on steve server

hughsheehy, 2022-04-27:
Probably a silly question. 

Is there any issue with connecting remotely to the Steve db and accessing the ongoing data from that db? 

Are there instructions anywhere on how to do it - if it doesn't cause any problems? 

H

goekay, 2022-04-27:
hey, sure it is possible. but the database (i.e. mysql or mariadb) that steve uses is a standalone resource and is separate from the java application itself (i.e. steve). therefore, this question has little to do with steve and a lot to do with database access/role management in general.

if your database is configured in a way to accept remote connections, sure, you can do this. read-only access should be pretty safe. on the other hand, if you start modifying data and tables, you might encounter weird consequences.

hughsheehy, 2022-04-28:
Cool. I kinda expected that would be the case. No plan to start modifying data tables....just read only.

goekay, 2022-05-03:
i hope the question was answered, and assuming the issue resolved.

closed 2022-05-03 by goekay