steve-community/steve #816: Unable to stop running transaction

lategoodbye, 2022-05-02:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Debian GNU/Linux 9 (stretch)
JDK               : 11
Database          : MySQL 5.7.29
```

#### Expected Behavior

It's possible to stop a running transaction with the stop button on the transaction overview.

#### Actual Behavior

Steve triggers an exception:
````
[INFO ] 2022-05-02 11:56:00,461 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Received: [2,"8e35819e1bb7824b0e4bded24f91618d0e9d","Authorize",{"idTag":"1ec11c60"}]
[INFO ] 2022-05-02 11:56:00,538 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Sending: [3,"8e35819e1bb7824b0e4bded24f91618d0e9d",{"idTagInfo":{"status":"Accepted","expiryDate":"2022-05-02T10:56:00.538Z"}}]
[INFO ] 2022-05-02 11:56:01,481 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Received: [2,"50566871f875c59fb9c341125c1e6c58469e","StartTransaction",{"connectorId":1,"idTag":"1ec11c60","meterStart":936,"timestamp":"2022-05-02T09:56:01.275Z"}]
[WARN ] 2022-05-02 11:56:01,551 de.rwth.idsg.steve.service.OcppTagService - The user with idTag '1ec11c60' is ALREADY in another transaction(s).
[INFO ] 2022-05-02 11:56:01,574 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Sending: [3,"50566871f875c59fb9c341125c1e6c58469e",{"transactionId":2644,"idTagInfo":{"status":"ConcurrentTx","expiryDate":"2022-05-02T10:56:01.552Z"}}]
[INFO ] 2022-05-02 11:56:02,613 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Received: [2,"b11c100a9f66922a248d383458f4497d60e3","StopTransaction",{"meterStop":936,"reason":"DeAuthorized","timestamp":"2022-05-02T09:56:02.483Z","transactionData":[{"sampledValue":[{"context":"Transaction.End","format":"SignedData","measurand":"Energy.Active.Import.Register","unit":"kWh","value":"4f434d467c7b224656223a22312e30222c224749223a22636861726765506f696e7456656e646f7220636861726765506f696e744d6f64656c222c224753223a2231323334353637383930222c224756223a22312e302e36222c225047223a225431373230222c224d56223a2249736b7261222c224d4d223a22574d334d34222c224d53223a225734313134353936222c224d46223a22322e3033222c224953223a747275652c22494c223a2248454152534159222c224946223a5b22524649445f504c41494e222c224f4350505f41555448225d2c224954223a2249534f3134343433222c224944223a223165633131633630222c224354223a22455653454944222c224349223a2244452a494e542a45434831323334353637383930222c225244223a5b7b22544d223a22323032322d30352d30325430393a35363a30302c3030302b303030302049222c225458223a2242222c225256223a302e39332c225249223a22312d623a312e382e30222c225255223a226b5768222c225254223a224143222c225354223a2247227d2c7b22544d223a22323032322d30352d30325430393a35363a30312c3030302b303030302049222c225458223a2272222c225256223a302e39332c225249223a22312d623a312e382e30222c225255223a226b5768222c225254223a224143222c225354223a2247227d5d7d7c7b225344223a2233303435303232313030393735323361643036633234643065653363343537393835343938363565343762663961383762373234653332333135313135363534326535363537353334643032323030666636626138653837323463323534386665313665373965623235353461636531346661346331313030373430633266633663656535663030323436356637227d"}],"timestamp":"2022-05-02T09:56:02.000Z"}],"transactionId":2644}]
[INFO ] 2022-05-02 11:56:02,630 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Sending: [3,"b11c100a9f66922a248d383458f4497d60e3",{}]
[INFO ] 2022-05-02 11:56:07,423 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Received: [2,"a23806fe4d499b3b0ff38a438d1e5c82b582","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing"}]
[INFO ] 2022-05-02 11:56:07,428 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=lesbos_left, sessionId=d450611f-8be4-4b97-5594-6f433bbdef2b] Sending: [3,"a23806fe4d499b3b0ff38a438d1e5c82b582",{}]
[ERROR] 2022-05-02 11:56:57,717 de.rwth.idsg.steve.web.GlobalControllerAdvice - Request: http://192.168.1.5:8081/steve/manager/transactions/stop/2642 raised following exception.
java.lang.NumberFormatException: For input string: "4f434d467c7b224656223a22312e30222c224749223a22636861726765506f696e7456656e646f7220636861726765506f696e744d6f64656c222c224753223a2231323334353637383930222c224756223a22312e302e35222c225047223a225431373136222c224d56223a2249736b7261222c224d4d223a22574d334d34222c224d53223a225734313134353936222c224d46223a22322e3033222c224953223a747275652c22494c223a2248454152534159222c224946223a5b22524649445f504c41494e222c224f4350505f4341434845225d2c224954223a2249534f3134343433222c224944223a223165633131633630222c224354223a22455653454944222c224349223a2244452a494e542a45434831323334353637383930222c225244223a5b7b22544d223a22323032322d30352d30325430363a31363a34352c3030302b303030302055222c225458223a2242222c225256223a302e39332c224546223a2274222c225249223a22312d623a312e382e30222c225255223a226b5768222c225254223a224143222c225354223a2247227d2c7b22544d223a22323032322d30352d30325430363a31373a32342c3030302b303030302055222c225458223a2272222c225256223a302e39332c224546223a2274222c225249223a22312d623a312e382e30222c225255223a226b5768222c225254223a224143222c225354223a2247227d5d7d7c7b225344223a2233303435303232303065316534623235633761646163376533343735366634343862343131383733353437653632396261613362636130636239663738316630386437323261333630323231303061346662613936636665663736313731386535626234613239333065363638666464333239616164613365383737613039646564313562346638336365363163227d"
        at jdk.internal.math.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2054) ~[?:?]
        at jdk.internal.math.FloatingDecimal.parseDouble(FloatingDecimal.java:110) ~[?:?]
        at java.lang.Double.parseDouble(Double.java:543) ~[?:?]
        at de.rwth.idsg.steve.service.TransactionStopService.findLastMeterValue(TransactionStopService.java:140) ~[steve.jar:?]
        at de.rwth.idsg.steve.service.TransactionStopService.findNeededValues(TransactionStopService.java:86) ~[steve.jar:?]
        at de.rwth.idsg.steve.service.TransactionStopService.stop(TransactionStopService.java:65) ~[steve.jar:?]
        at de.rwth.idsg.steve.web.controller.TransactionsReservationsController.stopTransaction(TransactionsReservationsController.java:87) ~[steve.jar:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.18.jar:5.3.18]
        at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.18.jar:5.3.18]
        at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.18.jar:5.3.18]
        at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.18.jar:5.3.18]
        at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.18.jar:5.3.18]
        at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.3.18.jar:5.3.18]
        at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1067) ~[spring-webmvc-5.3.18.jar:5.3.18]
[... cut ...]
        at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1038) ~[jetty-util-10.0.7.jar:10.0.7]
        at java.lang.Thread.run(Thread.java:834) ~[?:?]
````

#### Steps to Reproduce the Problem

Not sure how to reproduce this corner case, but it seems it has some to do with the stop transaction in the log before.
  1. start a transaction 2642
  2. try to start another transaction with the same RFID tag
  3. second get rejected and charging station sends stop transaction with OCMF in transaction data
  4. try to stop original transaction 2642

goekay, 2022-05-03:
hey @lategoodbye can you please post the the contents of the message from step 3?

lategoodbye, 2022-05-03:
Just an additional note, the chargepoint only send the OCMF in the stop transaction (not in MeterValues.Req). So it must come from a StopTransaction.Req.

goekay, 2022-05-03:
thanks @lategoodbye. there was a bug in the logic where we try to derive the stop value of a transaction from intermediate meter values. in your case the value was not Raw (which the logic was assuming) but SignedData.

fix is coming. thanks for raising this issue!

lategoodbye, 2022-05-04:
I tested the branch and the issue isn't reproducible anymore. Thanks a lot

goekay, 2022-05-04:
that's great. i merged the PR.

closed 2022-05-04 by goekay