steve-community/steve #817: No session context for chargeBoxId

followthebeast, 2022-05-02:
Hello Steve Community,

I have a site with nearly 500 DC chargers connected on Steve.

So we automated a TriggerMessage to see the status of the chargers every 2 hours.

What happened is that sometimes chargers tends to disconnect without a reason. We can see their active transactions and the status when changed but we can't send OCPP messages and for example the TriggerMessage to see the status (see logs below) :

[ERROR] 2022-05-02 08:45:05,380 de.rwth.idsg.steve.ocpp.ws.ChargePointServiceInvoker - Exception occurred
de.rwth.idsg.steve.SteveException: **No session context for chargeBoxId 'XXX-412'**
	at de.rwth.idsg.steve.ocpp.ws.SessionContextStoreImpl.getSession(SessionContextStoreImpl.java:129) ~[scb-steve-sa-current.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.getSession(AbstractWebSocketEndpoint.java:207) ~[scb-steve-sa-current.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ChargePointServiceInvoker.run(ChargePointServiceInvoker.java:81) ~[scb-steve-sa-current.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ChargePointServiceInvoker.runPipeline(ChargePointServiceInvoker.java:54) ~[scb-steve-sa-current.jar:?]
	at de.rwth.idsg.steve.ocpp.ChargePointService16_InvokerImpl.runPipeline(ChargePointService16_InvokerImpl.java:222) ~[scb-steve-sa-current.jar:?]
	at de.rwth.idsg.steve.ocpp.ChargePointService16_InvokerImpl.triggerMessage(ChargePointService16_InvokerImpl.java:81) ~[scb-steve-sa-current.jar:?]
	at de.rwth.idsg.steve.service.ChargePointService16_Client.lambda$triggerMessage$0(ChargePointService16_Client.java:87) ~[scb-steve-sa-current.jar:?]
	at java.util.ArrayList.forEach(ArrayList.java:1540) ~[?:?]
	at de.rwth.idsg.steve.service.BackgroundService$BackgroundListRunner.lambda$execute$0(BackgroundService.java:69) ~[scb-steve-sa-current.jar:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:834) [?:?]
Caused by: java.util.NoSuchElementException
	at de.rwth.idsg.steve.ocpp.ws.SessionContextStoreImpl.getSession(SessionContextStoreImpl.java:125) ~[scb-steve-sa-current.jar:?]
	... 14 more

Does anyone had this issue before ?

Thanks in advance.

Kind regards,

goekay, 2022-05-02:
> So we automated a TriggerMessage to see the status of the chargers every 2 hours.

steve shows only the ws/json stations that are connected at that moment on the Operations sub-pages. your automation probably bypasses this mechanism. 

if you try to send a message to a chargeboxid that is not connected at the moment, steve will not have any session (meaning: active/open websocket connection) for this chargeboxid. and hence the exception. i think the exception is justified. and there is no logic along the lines of "lets put this message in a queue to be sent later when the chargeboxid connects again".

followthebeast, 2022-05-02:
Thank you for your answer @goekay . But as i said, what we don't get is the charge box sends meter values and we can see the active transaction. But somehow, this charger is not in the list of "Number of connected json charge points" so it's normal that we can't do any operation.
Last question, how many chargers can Steve support (because as i told you we have nearly 500 chargers connected at the same time to Steve)

Thank you very much :) !

goekay, 2022-05-02:
> But as i said, what we don't get is the charge box sends meter values and we can see the active transaction.

imagine the following timeline:

t0: the station connects.
t1: a transaction starts, stations sends a StartTransaction -> the info is persisted in steve's db.
t2: meter values are sent for the transaction from t1 -> the info is persisted in steve's db.
t3: the station disconnects.
t4: you try to trigger an action which fails.
t5: station reconnects.
t6: the transaction from t1 stops, station sends a StopTransaction -> the transaction info is updated in steve's db.

steve reads the transaction data and meter values from database. if these were sent already, then no open connection is necessary at the moment of reading. after t1 you will start seeing this active transaction until t6. whether a transaction is active or not depends only on the fact whether a StopTransaction arrived or not. therefore, the transaction can remain active without an open connection even after t4 until t6.

in general: there might be time windows where a connection is open, and then time windows where the connection does not exist. if you are unlucky, you will encounter the exception (e.g. t4). 1 second later, the station might reconnect, and you would wonder why you had the exception. to better understand this lifecycle, you should consider the timestamps of the actions in addition to what these actions are.

> Last question, how many chargers can Steve support (because as i told you we have nearly 500 chargers connected at the same time to Steve)

this is hard to answer. i try to implement everything as efficiently as possible. but, everything i do comes from a theoretical perspective, because, disclaimer: i don't use steve in production, i do not operate charging stations. 

moreover, it depends on many factors: how many stations and how many users are there? how busy are the stations? how busy are the users? a better metric would be _operation/activity/message frequency_ on steve's end. you might have 10,000 stations that sit idle and do nothing. no problem. on the other hand, you might have 100 stations with 100 interactions each every second. different story.

another dimension is the resources you are running steve on: memory, cpu, database size, etc. steve is one single application with one single database behind it. this will apparently not scale and at some point will hit a wall, because too many things are happening in one box. but i don't where there this limit is.

i hope this helps.

followthebeast, 2022-05-03:
@goekay Thank you very much for this reply ! I will close the issue now.

closed 2022-05-03 by followthebeast

goekay, 2022-05-03:
@followthebeast can you please drop me an email. i would like to get to know your operation wrt. 500 stations better.