steve-community/steve #820: Alpitronic HyperCharger HYC 150 WS/Json not working after Update from 3.4.8 to 3.4.9

PeterPinnau, 2022-05-05:
SteVe Version     : 3.4.9
Operating system  : Ubutnu 20.04
JDK               :  OpenJDK 11.0.15
Database          : MariaDB 10.3.34

After Upgrade HyperCharger does not connect anymore. No heartbeat and no transaction can be send to SteVe. INFO log shows nothing. With loglevel DEBUG detailed messages appear. See attached log excerp for output of a connection attemp from 192.168.5.30 (HyperCharger) to SteVe 192.168.5.20
[steve_debug_log.txt](https://github.com/RWTH-i5-IDSG/steve/files/8630916/steve_debug_log.txt)

After downgrade to 3.4.8 communication works again.

goekay, 2022-05-05:
pls see https://github.com/RWTH-i5-IDSG/steve/issues/802 

i think you might have a similar case/problem.

PeterPinnau, 2022-05-05:
Thank you for feedback. URL in charger configuration also ends with '/' so I guess it will lead to '//', too. I will check this next week because the charger is in use at the moment and therefor configuration changes cannot be applied.
Unfortunately the URL is not logged so I cannot see if a double slash is present.

goekay, 2022-10-18:
closing due to inactivity.

closed 2022-10-18 by goekay