steve-community/steve #828: Charger Showing online for 15min after I switch off the charger

subhankar007b, 2022-05-20:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...3.4.9
Operating system  : ...Window 11
JDK               : ...11
Database          : ...Mysqk
```

#### Expected Behavior

Charger should show offline after I switch off charger

...

#### Actual Behavior

Charger should show online for 15min after I switch off charger

...

goekay, 2022-05-20:
if you do not gracefully shut down the json charger, it will not have the possibility to properly close the websocket connection. the information "charger just switched off" does not arrive at steve's end. how should it know this fact if no one communicates it? steve will continue thinking that the connection is there (even though it is _dangling_). steve will assess that charger is not accessible at most after 15 mins since we, by design, check the connections every 15 mins with ping-pong messages.

subhankar007b, 2022-05-20:
how to gracefully shut down the charger?
how to reduce 15min connection check to 5min?

goekay, 2022-05-20:
> how to gracefully shut down the charger?

no idea. specific to charger. this is out of scope of steve.

> how to reduce 15min connection check to 5min?

https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/config/WebSocketConfiguration.java#L54

you can change the value 15 in this line as desired. afterwards you need to rebuild and restart the project.

subhankar007b, 2022-05-20:
Ya i change it, but still its taking 15 min to close the connection.

subhankar007b, 2022-05-20:
[INFO ] 2022-05-20 16:07:30,972 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId='', sessionId=9388619c-cd03-bf88-d090-76da694c6b27] Sending: [3,"001014202205201837306221",{}]
[INFO ] 2022-05-20 16:07:31,027 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=, sessionId=9388619c-cd03-bf88-d090-76da694c6b27] Received: [
    2,
    "000008202205201837308192",
    "Heartbeat",
    {
    }
]

[INFO ] 2022-05-20 16:07:31,052 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=, sessionId=9388619c-cd03-bf88-d090-76da694c6b27] Sending: [3,"000008202205201837308192",{"currentTime":"2022-05-20T16:07:31.031+05:30"}]
[ERROR] 2022-05-20 16:28:54,866 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=, sessionId=9388619c-cd03-bf88-d090-76da694c6b27] Transport error
java.nio.channels.ClosedChannelException: null
	at org.eclipse.jetty.websocket.core.internal.WebSocketSessionState.onEof(WebSocketSessionState.java:169) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession.onEof(WebSocketCoreSession.java:250) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.fillAndParse(WebSocketConnection.java:485) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFillable(WebSocketConnection.java:343) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:319) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:100) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.SocketChannelEndPoint$1.run(SocketChannelEndPoint.java:101) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:894) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1038) ~[jetty-util-10.0.7.jar:10.0.7]
	at java.lang.Thread.run(Thread.java:834) ~[?:?]
[WARN ] 2022-05-20 16:28:54,867 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=, sessionId=9388619c-cd03-bf88-d090-76da694c6b27] Connection is closed, status: CloseStatus[code=1006, reason=Session Closed]

subhankar007b, 2022-05-20:
I'm getting this

subhankar007b, 2022-05-20:
If the charger switches off then it will break the WebSocket connection with steve right? and a ping pong message is sent when the charger is live and not doing any operation to check the connection?

goekay, 2022-05-20:
the logs you posted indicate that steve tries to ping the charger using the stale connection at 2022-05-20 16:28:54,866 but fails and decides that the connection is closed.

> If the charger switches off then it will break the WebSocket connection with steve right? 

yes but only in cases of graceful shutdowns. there is a ceremony between the 2 parties when closing a connection. if you unplug the charger, or something like that, this ceremony cannot be executed. then, steve will know that the connection was closed only during the next scheduled ping-pong after a while.