steve-community/steve #831: Rename or shorten URLs

mosea3, 2022-05-25:
#### Describe the problem you are trying to solve
My Wallbox (EN Plus) has sadly a very short memory for the OCPP Path.
So /steve/websocket/CentralSystemService/<Chargebox ID> is cut down to  /steve/websocket/CentralSystemServ/<ChargeboxID>

#### Describe the solution you'd like

Thus I'd like to rename the API path, 
I suggest /steve/ws/CSS/<Chargebox ID>

goekay, 2022-05-25:
i wouldn't want to change the defaults for one charging station.

you can change it as you wish though: https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/config/WebSocketConfiguration.java#L53

remember to rebuild and restart the application afterwards.

mhei, 2022-05-28:
It would be good to know what is the maximum string length of your wallbox. You could install SteVe without the /steve prefix, this would give you already 6 chars.

mosea3, 2022-05-30:
Hi guys,
Thank you very much for getting back to me.
As I see from the log
`[WARN ] 2022-05-20 18:29:50,832 org.springframework.web.servlet.PageNotFound - No mapping for GET /steve/websocket/SystemCentralSe/SNX`
I guess I suddenly since the firmware update I am 5 characters too short.

So @mhe s approach would be indeed preferrable.

As I don't have to change WebSocketConfiguration.java#L53 then, I simply change the context.path in main properties?
https://github.com/RWTH-i5-IDSG/steve/blob/5a8b68844b5595f91e521473af4d43258e1f9bcc/src/main/resources/config/prod/main.properties#L5
 Thanks!

goekay, 2022-05-30:
> As I don't have to change WebSocketConfiguration.java#L53 then, I simply change the context.path in main properties?

that's correct.

mosea3, 2022-06-03:
Thanks! 
for anyone finding this: I changed that line to
`context.path=`
(no value, empty, new line just after the = sign)

closed 2022-06-03 by mosea3