steve-community/steve #832: TriggerMessage: should also accept connector ID '0' for StatusNotification

barsnick, 2022-05-30:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9 + git master rev. 5a8b68844b5595f91e521473af4d43258e1f9bcc
Operating system  : Linux x86_64 (Fedora 35, kernel 5.17.11-300.fc36.x86_64)
JDK               : openjdk version "17.0.3" 2022-04-19, OpenJDK Runtime Environment 21.9 (build 17.0.3+7)
Database          : mariadb-10.5.15
```

#### Expected Behavior

Sends TriggerMessage with "requestedMessage" = "StatusNotification" and "connectorId" = 0.

#### Actual Behavior

Web interface refuses to send TriggerMessage, note "Connector ID must be at least 1".

#### Steps to Reproduce the Problem

  1. Select Operation -> OCPP v1.6
  2. Select operation TriggerMessage
  3. Choose Trigger Message: StatusNotification
  4. Try to set Connector ID to '0'

#### Additional context

The OCPP 1.6 standard says:
> For every such requested message the Central System MAY optionally indicate to which connector this request applies.

For StatusNotifications, connector ID '0' is obviously valid:
> Id '0' (zero) is used if the status is for the Charge Point main controller.

If the connector ID is omitted, a charging station reports StatusNotification for all valid connectors (i.e., 0, 1, and possibly further), which may not be what the Central System requires.

goekay, 2022-05-30:
> If the connector ID is omitted, a charging station reports StatusNotification for all valid connectors (i.e., 0, 1, and possibly further), which may not be what the Central System requires.

i agree that it's a little bit cumbersome, but the spec "ocpp-1.6 edition 2" only allows connector ids greater than 0 for TriggerMessage:
[image]

StatusNotification is only one of the many operations that can be triggered. connectorId0=0 is a special value for StatusNotification.

barsnick, 2022-05-31:
Ah, okay, indeed, TriggerMessage does not support this. Well, d'uh! I stand corrected.

Thanks!

closed 2022-05-31 by barsnick

goekay, 2022-05-31:
if you still want to change it for your setup, here is the relevant line in code: https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/dto/ocpp/TriggerMessageParams.java#L38

do not forget to recompile/rebuild and restart the service.