steve-community/steve #837: org.jooq.exception.DataAccessException when adding a Unknown Charge Point with ChargeBox ID starting by a space

juherr, 2022-06-08:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Linux
JDK               : 17
Database          : MySQL 8.0
```

#### Expected Behavior

The unknown charge point is added.

#### Actual Behavior

[image]

#### Steps to Reproduce the Problem

  1. Create a new charge point with ChargeBox ID (ie: `TEST`)
  2. Create a new unknown charge point with the previous ChargeBox ID starting by a space (ie: `%20TEST`)
  3. Add the unknown charge point 

#### Additional context

...

goekay, 2022-06-08:
this is because `String` fields go through `org.springframework.beans.propertyeditors.StringTrimmerEditor` as registered [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/GlobalControllerAdvice.java#L47) in order to sanitize and clean the input within the web/controller layer. this editor internally calls `String.trim()` which removes leading and trailing spaces.

i can understand the confusion because _what you see is not what you get_ but some input sanitation is necessary, don't you think? suggestion?

juherr, 2022-06-10:
> some input sanitation is necessary, don't you think? suggestion?

Yes, I totally agree it is often good that human inputs are sanitized.
I think it is also part of the business logic and could be better placed near the business code.

Space is a legit char for chargebox id and I didn't find any exceptions for the first or last char.
[image]

I think the design issue here and #836 is the inconsistency in the way the chargebox id is managed between the ocpp part and the UI part.

My suggestion:
* if space is a good first/last char then it should be managed in the whole application
* if space is not a good first/last char then it should not be shown as an Unknown Charge Point (maybe HTTP 400 on the websocket connection).

WDYT?

juherr, 2022-06-10:
> this is because String fields go through org.springframework.beans.propertyeditors.StringTrimmerEditor as registered [...]

Does it mean passwords are trimmed too?

goekay, 2022-06-16:
> Does it mean passwords are trimmed too?

which passwords (plural) you mean? the sign in credentials of the web ui are not affected by this.

juherr, 2022-06-17:
If password is kept as it, why not chargebox_id?
At least, when it comes from a station event.