steve-community/steve #845: Steve cannot be started via docker - reason unknown

arpekkar, 2022-06-22:
#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : MacOS
```

#### Expected Behavior

...

#### Actual Behavior

app_1  | [INFO] Building jar: /code/target/steve.jar
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] BUILD SUCCESS
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] Total time:  05:04 min
app_1  | [INFO] Finished at: 2022-06-22T08:23:39Z
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | Log file: /root/logs/steve.log
app_1  | Starting............ FAILED!
app_1  | Please refer to the log file for details

_steve.log_

[INFO ] 2022-06-21 12:56:55,564 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2022-06-21 12:56:55,721 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2022-06-21 12:56:55,817 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2022-06-21T12:56:55.765Z
[INFO ] 2022-06-21 12:56:56,991 org.eclipse.jetty.server.Server - jetty-10.0.7; built: 2021-10-06T19:34:02.766Z; git: da8a4553af9dd84080931fa0f8c678cd2d60f3d9; jvm 11.0.4+11
[INFO ] 2022-06-21 12:56:57,682 org.eclipse.jetty.server.handler.ContextHandler.steve - No Spring WebApplicationInitializer types detected on classpath
[INFO ] 2022-06-21 12:56:57,825 org.eclipse.jetty.server.session.DefaultSessionIdManager - Session workerName=node0
[INFO ] 2022-06-21 12:56:57,871 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring root WebApplicationContext
[INFO ] 2022-06-21 12:56:57,871 org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization started
[INFO ] 2022-06-21 12:56:59,925 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting..

#### Steps to Reproduce the Problem

1) change docker/main.properties

# Database configuration
#
...
db.user = me
db.password = password1234

# Credentials for Web interface access
#
auth.user = admin
auth.password = password1234

2) change docker-compose.yaml

  db:
    image: mariadb:10.4
    ports:
      - 3306:3306
    environment:
      MYSQL_RANDOM_ROOT_PASSWORD: "password1234"
      MYSQL_DATABASE: stevedb
      MYSQL_USER: me
      MYSQL_PASSWORD: password1234

3) docker-compose up

Mohit7497, 2022-07-19:
Hello @arpekkar, I am also having the same issue, did you get the solution?

keeth, 2023-01-14:
If there's no error in the log, it could be an out-of-memory error.

closed 2023-08-20 by goekay