steve-community/steve #848: Infinite message retry from station when the transactionId in MeterValues is invalid

juherr, 2022-06-23:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
Charge Point Vendor: Alfen BV
Charge Point Model: NG910-60005
Firmware Version: 4.8.0-3168
```

As #740, the `transactionId` from a `MeterValues` may not be related to an existing transaction.

#### Expected Behavior

The MeterValues message is stored even if it is not related to a known transaction.
Many solutions are possible:
* drop the fk between `connector_meter_value` and `transaction``
* as nullable, remove the `transactionId` when invalid (impossible value or not existing value in database)
* store the message in a dedicated table

#### Actual Behavior

The station is sending at infinite the same message that will never be ack due to the exception.
This message is blocking the followers and clearing the memory cache remotely may not work.

#### Steps to Reproduce the Problem

  1. Sending a MeterValues with an invalid `transactionId`

#### Additional context

Stacktrace:
```
[INFO ] 2022-06-23 20:32:15,641 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=XXXX, sessionId=309c2c35] Received: [2,"10739","MeterValues",{"connectorId":1,"transactionId":-1,"meterValue":[{"timestamp":"2020-10-28T15:08:41Z","sampledValue":[{"value":"0.010","context":"Transaction.End","unit":"kWh"}]}]}]
[ERROR] 2022-06-23 20:32:15,644 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint$Ocpp16CallHandler - Exception occurred
org.jooq.exception.DataAccessException: SQL [insert into `xxxx`.`connector_meter_value` (`connector_pk`, `transaction_pk`, `value_timestamp`, `value`, `reading_context`, `format`, `measurand`, `location`, `unit`, `phase`) values (?, ?, ?, ?, ?, ?, ?, ?, ?, ?)]; Data truncation: Out of range value for column 'transaction_pk' at row 1
	at org.jooq_3.11.4.MYSQL.debug(Unknown Source) ~[?:?]
[...]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.batchInsertMeterValues(OcppServerRepositoryImpl.java:424) ~[steve.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.lambda$insertMeterValues$1(OcppServerRepositoryImpl.java:142) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext$3.run(DefaultDSLContext.java:584) ~[steve.jar:?]
[...]
	at org.jooq.impl.DefaultDSLContext.transaction(DefaultDSLContext.java:581) ~[steve.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.insertMeterValues(OcppServerRepositoryImpl.java:137) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.CentralSystemService16_Service.meterValues(CentralSystemService16_Service.java:127) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.CentralSystemService16_SoapServer.meterValues(CentralSystemService16_SoapServer.java:83) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint$Ocpp16CallHandler.dispatch(Ocpp16WebSocketEndpoint.java:70) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.AbstractCallHandler.accept(AbstractCallHandler.java:28) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:43) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:90) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleMessage(AbstractWebSocketEndpoint.java:62) ~[steve.jar:?]
	at org.springframework.web.socket.adapter.jetty.JettyWebSocketHandlerAdapter.onWebSocketText(JettyWebSocketHandlerAdapter.java:83) ~[steve.jar:?]
	at jdk.internal.reflect.GeneratedMethodAccessor79.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72) ~[steve.jar:?]
[...]
 	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:680) ~[steve.jar:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
Caused by: java.sql.BatchUpdateException: Data truncation: Out of range value for column 'transaction_pk' at row 1
	at jdk.internal.reflect.GeneratedConstructorAccessor1468.newInstance(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45) ~[?:?]
	at java.lang.reflect.Constructor.newInstance(Constructor.java:490) ~[?:?]
	at com.mysql.cj.util.Util.handleNewInstance(Util.java:210) ~[steve.jar:?]
[...]
	at com.mysql.cj.jdbc.StatementImpl.executeBatch(StatementImpl.java:813) ~[steve.jar:?]
	at com.zaxxer.hikari.pool.ProxyStatement.executeBatch(ProxyStatement.java:128) ~[steve.jar:?]
	at com.zaxxer.hikari.pool.HikariProxyPreparedStatement.executeBatch(HikariProxyPreparedStatement.java) ~[steve.jar:?]
	at org.jooq.tools.jdbc.DefaultStatement.executeBatch(DefaultStatement.java:100) ~[steve.jar:?]
	at org.jooq.impl.BatchSingle.executePrepared(BatchSingle.java:237) ~[steve.jar:?]
	... 50 more
Caused by: com.mysql.cj.jdbc.exceptions.MysqlDataTruncation: Data truncation: Out of range value for column 'transaction_pk' at row 1
	at com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:104) ~[steve.jar:?]
[...]
	at com.mysql.cj.jdbc.StatementImpl.executeBatch(StatementImpl.java:813) ~[steve.jar:?]
	at com.zaxxer.hikari.pool.ProxyStatement.executeBatch(ProxyStatement.java:128) ~[steve.jar:?]
	at com.zaxxer.hikari.pool.HikariProxyPreparedStatement.executeBatch(HikariProxyPreparedStatement.java) ~[steve.jar:?]
	at org.jooq.tools.jdbc.DefaultStatement.executeBatch(DefaultStatement.java:100) ~[steve.jar:?]
	at org.jooq.impl.BatchSingle.executePrepared(BatchSingle.java:237) ~[steve.jar:?]
	... 50 more
```

It is clearly a station issue but it is easier to make a soft workaround than fixing hardware.
Another

goekay, 2022-06-24:
what about catching and logging the exception as a solution? i am reluctant to change the database schema and endanger data integrity.

goekay, 2022-06-24:
> what about catching and logging the exception as a solution?

actually, after taking a look at the respective part in the code, this is already the behaviour. any exception that `batchInsertMeterValues` might throw is being catched.

are you sure that there is no ack?

juherr, 2022-06-25:
Sorry, I was not precise enough: the server replies as expected `Sending: [4,"xx","InternalError","Internal services failed while processing of the payload",{"errorMsg":"SQL [insert into `pdbrs`.`connector_meter_value` (`connector_pk`, `transaction_pk`, `value_timestamp..."}`. Then the station is retrying to send the exact same message a few time later again and again.

I have a similar issue with another station but with StopTransaction instead.

In fact, I remember how to reproduce the issue easily: 
* start a transaction
* change the station configuration and move it to another server
* stop the transaction

You'll never receive the StopTransaction message.

It is not a steve issue but not a station issue either. It is a human issue.

Is it possible that steve manages such bad behavior?

lategoodbye, 2022-06-27:
> Sorry, I was not precise enough: the server replies as expected `Sending: [4,"xx","InternalError","Internal services failed while processing of the payload",{"errorMsg":"SQL [insert into [...]

Quote from OCPP 1.6 spec:
> It is likely that The Central System applies sanity checks to the data contained in a MeterValues.req it
received. The outcome of such sanity checks SHOULD NOT ever cause the Central System to not
respond with a MeterValues.conf

So in my point of view this is a Steve issue.

goekay, 2022-06-27:
let's focus on one thing, because i have a feeling we are talking about multiple flows. MeterValues req/res flow is different from StartTransaction req/res or StopTransaction req/res flow.

MeterValues req/res flow should not "not respond" to a request. yes, it will log the exception and the whole stacktrace. yes, it will be a lot of stuff. but at the end, you should see a response message with the same id in the logs.

business logic is [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/CentralSystemService16_Service.java#L153-L161). the database layer is [here](https://github.com/RWTH-i5-IDSG/steve/blob/a00b490b1c5b4dc920fa486ec49f0d34067bc619/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java#L169-L209).

as you can see, both places in database layer use steps that are wrapped in a try-catch block. we catch every exception and log it. therefore, the exception will not bubble up to CentralSystemService16_Service. therefore, the next line where the response object is constructed and returned will be executed.

i could verify and validate these things in my local tests just now. the client always receives a response.

juherr, 2022-06-27:
## Expected flow

```mermaid
sequenceDiagram
    participant S as Station
    participant CS as CentralSystem
    S->>+CS: req.StartTransaction
    CS-->>-S: res.StartTransaction
    S->>+CS: req.MeterValues
    CS-->>-S: res.MeterValues
    S->>+CS: req.StopTransaction
    CS-->>-S: res.StopTransaction
```

## Actual flow when Transaction is missing (if the Station was configured to another CentralSystem during the Transaction, or if the Station was removed/recreate from the CentralSystem during the Transaction)

### Case 1: MeterValues

```mermaid
sequenceDiagram
    participant S as Station
    participant CS as CentralSystem
    participant CS2 as CentralSystem2
    S->>+CS: req.StartTransaction
    CS-->>-S: res.StartTransaction
    CS->>+S: req.UpdateConfiguration
Note left of S: CentralSystem changed
    S-->>-CS: res.UpdateConfiguration
loop
    S->>+CS2: req.MeterValues
Note right of CS2: Exception
    CS2-->>-S: res.MeterValues
end
```

### Case 2: StopTransaction

```mermaid
sequenceDiagram
    participant S as Station
    participant CS as CentralSystem
    participant CS2 as CentralSystem2
    S->>+CS: req.StartTransaction
    CS-->>-S: res.StartTransaction
    S->>+CS: req.MeterValues
    CS-->>-S: res.MeterValues
    CS->>+S: req.UpdateConfiguration
Note left of S: CentralSystem changed
    S-->>-CS: res.UpdateConfiguration
loop
    S->>+CS2: req.StopTransaction
Note right of CS2: Exception
    CS2-->>-S: res.StopTransaction
end
```

As I understand, case 1 is already supposed to work in the latest release.
What about case 2?

Fyi, I'm not able to determine the exact version because `/about` is failing and I've no access direct access to the server.
I'll move failing stations to an up-to-date server I can manage easily and I will try to see what is the behavior after that.

goekay, 2022-06-27:
both cases are and will be working. the exceptions will be there. they will be catched and logged. the station will receive a response regardless. this behaviour has been like this for a long time. [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/CentralSystemService16_Service.java#L199) is the part for the interested to follow the call chain.

in the second case, the regular stop transaction flow will fail to find the transaction id in database and therefore fail to insert it into the regular table, and will fallback to insert it to a `transaction_stop_failed` table without the referential constraints. if any meter values were part of the payload, steve will fail to insert them as well. but in this case there is no fallback. steve will just log them.

juherr, 2022-06-28:
Ok. It looks like my issue is a duplicate of #156

It means my instance is very older than what I expected.

Sadly, I lost the communication with the station when I tried to move to another instance and I will have to go there.

I close the issue and maybe will reopen it if I will have more details.

Thanks for the time you took to look for and sorry for the waste of it.

closed 2022-06-28 by juherr