steve-community/steve #858: Since last release:  Unrecognized field "Reason" (class ocpp.cs._2015._10.StopTransactionRequest), not marked as ignorable (6 known properties: "reason", "timestamp", "transactionData", "transactionId", "meterStop", "idTag"])

mosea3, 2022-07-12:
Hi guys,
Since last release I seem to get the following error, when im stopping any Transaction:
 Unrecognized field "Reason" (class ocpp.cs._2015._10.StopTransactionRequest), not marked as ignorable (6 known properties: "reason", "timestamp", "transactionData", "transactionId", "meterStop", "idTag"])

I think its even just a typo "Reason" instead of "reason", but I dont get to effectively change it in the relevant classes.

On a previous release 3.4.8, I dont have this issue.

#### Checklist

- [x] I checked other issues already, but found no answer/solution:
similar parsing error, but after firmware update and not transaction related.
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version         : 3.4.9
Operating system  : Ubuntu 22.04 LTS
JDK                         : OpenJDK 18 "ea"
Database                : MariaDB 10.6.7 
```

`[ERROR] 2022-07-12 07:59:40,897 de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer - Exception occurred
com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "Reason" (class ocpp.cs._2015._10.StopTransactionRequest), not marked as ignorable (6 known properties: "reason", "timestamp", "transactionData", "transactionId", "meterStop", "idTag"])
 at [Source: UNKNOWN; byte offset: #UNKNOWN] (through reference chain: ocpp.cs._2015._10.StopTransactionRequest["Reason"])
	at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:61) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:1127) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.deser.std.StdDeserializer.handleUnknownProperty(StdDeserializer.java:2023) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperty(BeanDeserializerBase.java:1700) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownVanilla(BeanDeserializerBase.java:1678) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:319) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:176) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:323) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.ObjectMapper._readValue(ObjectMapper.java:4650) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2831) ~[jackson-databind-2.13.3.jar:2.13.3]
	at com.fasterxml.jackson.databind.ObjectMapper.treeToValue(ObjectMapper.java:3295) ~[jackson-databind-2.13.3.jar:2.13.3]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.handleCall(Deserializer.java:128) ~[steve-release.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.accept(Deserializer.java:78) ~[steve-release.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:49) ~[steve-release.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:115) ~[steve-release.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.onMessage(AbstractWebSocketEndpoint.java:87) ~[steve-release.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ConcurrentWebSocketHandler.handleMessage(ConcurrentWebSocketHandler.java:50) ~[steve-release.jar:?]
	at org.springframework.web.socket.adapter.jetty.Jetty10WebSocketHandlerAdapter.onWebSocketText(Jetty10WebSocketHandlerAdapter.java:78) ~[spring-websocket-5.3.20.jar:5.3.20]
	at org.eclipse.jetty.websocket.core.internal.messages.StringMessageSink.accept(StringMessageSink.java:53) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.acceptMessage(JettyWebSocketFrameHandler.java:334) ~[websocket-jetty-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.onTextFrame(JettyWebSocketFrameHandler.java:407) ~[websocket-jetty-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.onFrame(JettyWebSocketFrameHandler.java:240) ~[websocket-jetty-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession$IncomingAdaptor.lambda$onFrame$0(WebSocketCoreSession.java:660) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.server.handler.ContextHandler.handle(ContextHandler.java:1463) ~[jetty-server-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.server.handler.ContextHandler.handle(ContextHandler.java:1482) ~[jetty-server-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.server.internal.AbstractHandshaker$1.handle(AbstractHandshaker.java:212) ~[websocket-core-server-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession$IncomingAdaptor.onFrame(WebSocketCoreSession.java:660) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.ExtensionStack.onFrame(ExtensionStack.java:101) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession.onFrame(WebSocketCoreSession.java:476) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFrame(WebSocketConnection.java:262) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.fillAndParse(WebSocketConnection.java:458) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFillable(WebSocketConnection.java:343) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:319) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:100) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.SocketChannelEndPoint$1.run(SocketChannelEndPoint.java:101) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.runTask(AdaptiveExecutionStrategy.java:412) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.consumeTask(AdaptiveExecutionStrategy.java:381) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.tryProduce(AdaptiveExecutionStrategy.java:268) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.lambda$new$0(AdaptiveExecutionStrategy.java:138) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:407) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:894) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1038) ~[jetty-util-10.0.7.jar:10.0.7]
	at java.lang.Thread.run(Thread.java:833) ~[?:?]
[INFO ] 2022-07-12 07:59:40,898 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SNX, sessionId=6e815dc7-4e78-5790-cbdb-9cdd20bb8b17] Sending: [4,"83359","FormationViolation","The payload for action could not be deserialized",{"errorMsg":"Unrecognized field \"Reason\" (class ocpp.cs._2015._10.StopTransactionRequest), not marked as ignorabl..."}]
[INFO ] 2022-07-12 07:59:51,590 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SNX, sessionId=6e815dc7-4e78-5790-cbdb-9cdd20bb8b17] Received: [2,"83360","StatusNotification",{"connectorId":1,"status":"Preparing","errorCode":"NoError","timestamp":"2022-07-12T07:59:50Z","vendorId":"EN+","info":"{\"reason\":\"plugInGun\",\"cpv\":0,\"rv\":0}","vendorErrorCode":"00000000000000000000000000000000"}]
[INFO ] 2022-07-12 07:59:51,593 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SNX, sessionId=6e815dc7-4e78-5790-cbdb-9cdd20bb8b17] Sending: [3,"83360",{}]
[INFO ] 2022-07-12 07:59:56,294 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SNX, sessionId=6e815dc7-4e78-5790-cbdb-9cdd20bb8b17] Received: [2,"83361","StopTransaction",{"meterStop":216975,"transactionId":23,"timestamp":"2022-07-12T07:59:08Z","idTag":"fc2cb183","Reason":"EVDisconnected"}]`

#### Expected Behavior
- to finish Transaction within steve

#### Actual Behavior
- Transaction does not stop, steve not ready to start a new transaction

mosea3, 2022-07-12:
any advice where to go and case-match the fields?

I already looked into those classes, but those generic fields seem to be defined somewhere else.

goekay, 2022-07-12:
after looking at the spec we realized that the JSON structures mirror the SOAP structures. there were official WSDL files (for SOAP) which describe the SOAP message structures. there are plugins to auto-gen Java classes from WSDL. and our [side project](https://github.com/RWTH-i5-IDSG/ocpp-jaxb) does exactly this. and we use the same Java classes from websocket stations as well.

long story short, the class `ocpp.cs._2015._10.StopTransactionRequest` and its fields are auto-generated based on the spec.

some time ago i implemented [this workaround](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/custom/MeterValue15Deserializer.java) for an analogous issue. maybe it can help you as an inspiration.

goekay, 2022-07-12:
on a more general note, are you sure about it working with 3.4.8? i just tried the following:

```
public static void main(String[] args) throws Exception {
    ocpp.cs._2015._10.StopTransactionRequest req = de.rwth.idsg.steve.ocpp.ws.JsonObjectMapper.INSTANCE.getMapper()
        .readValue(
            "{\"meterStop\":216975,\"transactionId\":23,\"timestamp\":\"2022-07-12T07:59:08Z\",\"idTag\":\"fc2cb183\",\"Reason\":\"EVDisconnected\"}",
            ocpp.cs._2015._10.StopTransactionRequest.class
        );

    System.out.println(req);
}
```

in code bases of both 3.4.8 and 3.4.9. both throw exceptions. if i convert `Reason` to `reason` both succeed.

this means that the deserialization logic/strictness did not change between these versions. i must assume that the issue lies within your setup.

mosea3, 2022-08-09:
Hi there,
I now built steve.jar on 2 different systems and copied it run it on the server, still gets me this issue

Where would I find the WSDLs in the project so that I can very strictly use exactly the same wording?
Thanks :)

goekay, 2022-08-09:
https://github.com/RWTH-i5-IDSG/ocpp-jaxb/tree/master/src/main/resources/wsdl

closed 2023-08-20 by goekay