steve-community/steve #869: Steve Configuration/Installation

KareemKelgamal, 2022-08-02:
Hi
This Kareem from Egypt
I used to follow the installation/config. steps and I downloaded all tools needed for system, then I prepared (steve.db) database but I have a misunderstanding concerning building and running of steve, if you can go in details in step 4 and 5 instructions cause i stuck at this point.

Thanks

mosea3, 2022-08-09:
its a Java project, with Maven package manager which builds it.
So you need to install mvn and its dependencies

Thats why you use command
`mvn package`

this will build it.
If it fails, run  mvn -X to see a more detailed version of building.

This will create in the folder ./steve/target/ a new executable Java file (JAR), steve.jar

To execute this manually, you will need to call the command like in the manual.
`java -jar steve.jar`
It then runs as long as the command is active.

To have it running in the background, you might want to run it as a service.
I in this case created a new "service unit" in
/etc/systemd/system/steve.service

```
[Unit]
Description=Runs SteVe JAR as Service
[Service]
WorkingDirectory=/home/pcm/steve/target
ExecStart=/usr/bin/java -jar steve-release.jar
ExecStop=/bin/kill -15 $MAINPID
SuccessExitStatus=143

Type=simple

User=<!type in your username here, which should execute it!>
Group=<!type in your users groupname here, which should execute it!>

[Install]
WantedBy=multi-user.target

```

and then we can use systemctl or service to run
`service steve start`
to start it
`service steve status`
to see the service status (eg. if the service loaded, or failed prematurely)
and
`service steve stop`
to stop the service

very convenient to run it in the background.
Logs are still there where you set it up in [log4j](https://github.com/RWTH-i5-IDSG/steve/blob/364ae967d0a7bafd7da8838a65a4d13dc9918dfd/src/main/resources/config/prod/log4j2.xml#L10)

Does that make it any clearer?
Or can you describe at which point youre stuck?

KareemKelgamal, 2022-08-10:
@mosea3 
much thanks

goekay, 2022-09-07:
i consider this issue to be resolved, and therefore am closing it.

closed 2022-09-07 by goekay