steve-community/steve #870: Bulk upload of OCPP tags

rdc-Green, 2022-08-02:
I'm trying to upload multiple OCPP tags. Will the ocpp_tg_pk automatically update for each new row added?

rdc-Green, 2022-08-10:
thank you @mosea3

closed 2022-08-10 by rdc-Green

fluppie, 2024-08-16:
@mosea3 is this: "I'd recommend importing from CSV over command line" explained somewhere? Or I'm I completely blind for not having found the instructions?

goekay, 2024-08-16:
> Steve has no bulk import function yet.

not quite true. if you go to ocpp tags page, and click on "Add New", you will be presented with a page that contains two sections: Add OCPP Tag (which is the default) and Add OCPP Tag List above it which an be toggled on. then you can paste into the text area your tags to be bulk-inserted ;)

[image]

fluppie, 2024-08-18:
How could I have missed that...
Is it hard to create the same for users?
At the moment I have a file with a name and a tag ID, coolest would be to import them in one go:
User1 00112233445566778899
User2 99887766554433221100