steve-community/steve #873: Charge Point Configuration

KareemKelgamal, 2022-08-10:
After setting Charge point and ChargeBoxID, I couldn't find it at operations so I can't select it for setting ConnecterID and other OCPP version.
[image]

goekay, 2022-08-10:
after registering a charging station at steve, you should also configure the station to communicate with steve.

the station informs steve about the connectors etc.

KareemKelgamal, 2022-08-11:
Thanks @goekay 
but I'm still working on system simulation and I use steve with OCPP simulator

goekay, 2022-08-11:
> I use steve with OCPP simulator

well, in that case, i don't know how to help you.

KareemKelgamal, 2022-08-11:
Ok Mr. @goekay 
I have few questions
1-how can i reset the data showed at steve, I can delete charge points and other information but OCPP tags remains!?
2-Is there any way to change the user & password for steve as after I made my configurations in main properties file, it still opens with the default user name and password.

goekay, 2022-08-11:
> 1-how can i reset the data showed at steve, I can delete charge points and other information but OCPP tags remains!?

please explore the pages under "data management". you can see/update/delete entities.

> 2-Is there any way to change the user & password for steve as after I made my configurations in main properties file, it still opens with the default user name and password.

if you change something within the properties file, you need to build and start the project again.

closed 2023-08-20 by goekay