steve-community/steve #882: Maven Issue

andle, 2022-08-24:
#### Checklist

- [ x] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9 (trying to build latest)
Operating system  : Raspbian GNU/Linux 10 (buster)
JDK               : 11.0.16
Database          : mariadb  Ver 15.1 Distrib 10.3.34-MariaDB
```

#### Expected Behavior
Build success

#### Actual Behavior
Errors

#### Steps to Reproduce the Problem

  1. Pulled latest code
  2. mvn package
  3. Errors

#### Additional context
Can someone help me understand this build error? I am not very good with maven. Pulled the latest today and it would not build.

 mvn package
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/share/maven/lib/guice.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[ERROR] Error executing Maven.
[ERROR] Unable to create injector, see the following errors:

1) No implementation for org.apache.maven.bridge.MavenRepositorySystem annotated with interface org.eclipse.sisu.inject.TypeArguments$Implicit was bound.
  Did you mean?
    org.apache.maven.bridge.MavenRepositorySystem bound  at org.eclipse.sisu.wire.LocatorWiring

  at org.eclipse.sisu.wire.LocatorWiring

1 error

goekay, 2022-09-07:
hey, i am not sure what these warnings and errors related to guice and eclipse should mean. where are you executing maven?

i can only imagine that you are executing maven within eclipse (IDE), and eclipse tries to do some voodoo but fails.

andle, 2022-09-07:
@goekay I am executing the mvn package build from bash. 
This is on a raspberry pi where I have not installed eclipse.

andle, 2022-09-13:
Not sure what changed, but it is working now.
Closing issue.

closed 2022-09-13 by andle