steve-community/steve #889: Error 400 invalid SNI https

dargoc2001, 2022-08-29:
#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Centos 7

Hello, I have been having problems to release the steve project as HTTPS with the 400 error code "Invalid SNI"
I created a keystore with the command **keytool -genkey -alias hostname -keyalg RSA -keystore keystore.jks**
In -alias hostname, the hostname is literally the name appearing in /etc/hostname
![image](https://user-images.githubusercontent.com/112375779/187141490-bac83f3f-4adc-48ad-bfae-69846ea551df.png)

I tried to resolve it several times but I couldn´t so Im writting this post.
Hope you can help me

goekay, 2022-09-07:
pls look at https://github.com/steve-community/steve/issues/749. looks like a similar/duplicate issue.

goekay, 2022-10-18:
closing due to inactivity.

closed 2022-10-18 by goekay