steve-community/steve #895: Community feedback wanted: Migrate from Java 11 to Java 17

goekay, 2022-09-03:
Hey all,

our Java version, Java 11, has been released in 2018. Java 17 has been released in 2021. Both are LTS. I want to migrate the project to Java 17 because there is some momentum behind it:

1. JooQ (our DB abstraction layer) made it the baseline for new versions [[source](https://blog.jooq.org/3-17-0-release-with-computed-columns-audit-columns-pattern-matching-reactive-transactions-and-kotlin-coroutine-support/)]. Because of this we cannot update it, merge automatic PRs from dependabot starting with [this](https://github.com/steve-community/steve/pull/853).
2. Next Spring version will be released in Q4 2022. Java 17 will be the baseline for Spring 6: https://spring.io/blog/2021/09/02/a-java-17-and-jakarta-ee-9-baseline-for-spring-framework-6

The cadence of new Java version releases increased in recent years. The migrations from an older to a newer release are less of an issue for users.

Feedback and ideas are welcome. Are you okay with this?

juherr, 2022-09-05:
A rule we use at @testng-team is: only need to support the 2 last LTS.
But TestNG is a library, not a project.

The latest LTS is supposed to be enough for a free hobby project. 
If someone can't use the latest Java LTS version, I suppose they can pay for its support.

Maybe a `features/java17-migration` branch with the changes could help to provide more feedback.

csamsel, 2022-09-12:
the latest raspian version is based on debian 11 which comes by default with java 11. Ubuntu LTS 20.04 which is also still quiet common as base image e.g. in Azure also comes with java 11 (22.04 comes with java 17).

So it would be some hassle for a lot of users. I would wait a bit more.

lategoodbye, 2022-09-21:
FWIW I successfully tested Steve 3.4.9 with Debian 11 and OpenJDK 17.0.4 (amd64), which is also available via apt.

goekay, 2022-11-20:
just as a note: spring 6 is out and PRs created by dependabot started failing. example: https://github.com/steve-community/steve/pull/971

keeth, 2023-01-14:
Upgrade to Java 17 sounds good to me!

closed 2024-01-06 by goekay