steve-community/steve #896: Community feedback wanted: Migrate from MySQL 5.7 to MySQL 8

goekay, 2022-09-03:
Hey all,

MySQL 5.7 has been released in late 2015. MySQL 8 is around since 2018. We support MySQL 5.7 and upwards. Our DB migration tool Flyway dropped MySQL 5.7 in its free tier. Therefore, we cannot upgrade Flyway beginning with [this](https://github.com/steve-community/steve/pull/676).

I want to make MySQL 8 the baseline. Feedback and ideas are welcome. Are you okay with this?

shadsnz, 2022-09-03:
Okay by me.

lategoodbye, 2022-09-04:
I understand the motivation. Unfortunately Debian based Linux distribution are relative conservative regarding new packages especially to databases, so MySQL 8 is included just in Ubuntu 22.04.

Is there some kind of timeline for the migration?
Will there be a step wise migration like a temporary test & development branch for the brave ones?

Personally i suggest to migrate to MySQL 8 before migrate to Java 17.

juherr, 2022-09-05:
@goekay What will be the impact? I think having a `features/mysql8-migration` with the changes (at least the jooq plugin upgrade) could help to give feedbacks.

goekay, 2022-09-05:
since we support mysql 8 already, there won't be any code changes. we will just be dropping the support of mysql 5.7. this means:
- updating the readme.
- removing the github actions builds that include mysql 5.7.
- update flyway's maven plugin.

this issue has more of a _"what if..."_ character and is a heads up, since we might break your installation if you still use mysql 5.7. i am trying to gauge community support or backlash.

lategoodbye, 2022-09-17:
Just to get this right, MariaDB (10.2 and higher) support is completely unaffected by this change?

goekay, 2022-10-18:
i am not sure about "10.2 and higher" but i think "10.3 and higher" should be fine. we have 10.3 and 10.5 in our github action build matrix and they pass.

closed 2024-01-06 by goekay