steve-community/steve #897: How to get the state of charge of charging vehicle in percentage in StartTransaction.req

shtaigen, 2022-09-05:
#### Checklist

- [X] I checked other issues already and found no answer
- [X] I checked the documentation and found no answer
- [X] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

I try to get Soc before a vehicle starts charging.

#### Describe the solution you'd like

I'd like to get Soc in StartTransaction.req or to get meterStart with the meter value not in Wh but in percentage

goekay, 2022-09-05:
the meterStart and meterStop in StartTransaction.req and StopTransaction.req, respectively, are readings from the internal meter of the charging station. they do not represent SoC of the vehicle.

however, meterValue in MeterValues.req and transactionData in StopTransaction.req contains a data structure with a field called "measurand" which is an enum and has SoC as one of the possible enum values.

so, technically, this is possible. but the charging station would need to get this information from somewhere. this depends on the intelligence of the station and the car and their communication.

goekay, 2022-09-05:
related: https://github.com/steve-community/steve/issues/532

closed 2022-09-07 by goekay