steve-community/steve #909: Problem Starting Steve behind a proxy server with docker

itsn0k, 2022-09-13:
#### Checklist

- [yes] issues already, but found no answer/solution
- [yes] I checked the documentation and wiki, but found no answer/solution
- [yes] I am running the latest version and the issue still occurs
- [no] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : Steve-3.4.9 docker container
Operating system  : Ubuntu 20.04 
JDK               : maven:3.6.1-jdk-11
Database          : mariadb:10.4
```

#### Expected Behavior
Docker application starts with in about 2-5 minutes 

How the logs usually look 
[image]

#### Actual Behavior
Docker application starts with in about 60-90 minutes and I can't connect any charge points.

When I look at the logs I notice that when I start Steve I run into an error that ends after 60-90 minutes with a timeout and then he starts the server successfully. But I can not connect any charge points to the server. 

I assume that it is related to the following log or short-circuited command: org.hibernate.validator.internal.util.Version (main) - HV000001: Hibernate Validator 6.1.7.Final

What the logs look like with the error:
[image]
after 60-90 minutes
[image]

#### Steps to Reproduce the Problem

  1. Server behind a proxy server with certificate type
  2. Configure Steve and server as specified 
  3. start steve with docker-compose up

#### Additional context
Modifications due to proxy server

1. Modifications to server _/etc/environment_  
[image]
2. Improving the proxy certificate _/usr/local/share/ca-certificates/ca.crt_
3. Adaptations to Docker for pull and push _nano /etc/systemd/system/docker.service.d/http-proxy.conf_
[image]
4. Adaptations to Docker so that Conatainer can use wget _.docker/config.json_ + Certificate import
[image]
5. Adaptation to Conatiner image so that I can use mvn _/usr/share/maven/conf/settings.xml_
[image]

csamsel, 2022-09-13:
the logfiles arent readable. Is there a reason you want to use docker and a proxy? It makes the setup a lot more complicated. 

dont forget you need to proxy a _websocket connection_ in both directions for ocpp-j

itsn0k, 2022-09-15:
Hi, 
Problem is that I can't get past the proxy when I want to get on the internet. As far as I understand, not using Docker would not change anything. Since the problem is when starting the Steve app. 

What exactly do you mean by „need to proxy a websocket connection in both directions for ocpp-j”

My understanding is that I only need the proxy to download the Steve app. Since Steve and the Charge Point are on the same network and both are already communicating with each other. Without going through the proxy server as it is only needed for an Internet connection.

keeth, 2023-01-14:
The Dockerfile in this repo has a number of problems.  Doing the entire build in the "run" stage is not a good pattern.

You're welcome to try the Dockerfile I made for Steve:

https://github.com/keeth/steve/blob/hv/Dockerfile

It uses a multi-stage build so that only the jar is copied to the final image.  I build it like this:

`docker build -t keeth/steve --network=host .`

The `--network=host` bit is because the build requires a connection to a mariadb instance that has been initialized with the Steve SQL.  Also I added `127.0.0.1 mariadb` to /etc/hosts on my laptop so that the `mariadb` hostname would resolve.

Hope this helps!

juherr, 2026-09-15:
This issue is one of the reasons I built an unofficial ready-to-run SteVe image:

https://github.com/juherr/steve-ocpp-csms-image

SteVe is compiled when the image is built, so container startup does not run Maven or download dependencies. That should make deployments behind restrictive proxies much more predictable.

Feedback from this kind of environment would be very useful.