steve-community/steve #910: API endpoints for automation and integration

kohlerdominik, 2022-09-17:
#### Describe the problem you are trying to solve

We are trying to interact with SteVe with a third party service in a micro service architecture. SteVe should manage the charging points, but we need to extract consumed energy information.

#### Describe the solution you'd like

If SteVe would provide API-endpoints for relevant information, Machine-to-Machine communication and therefore automation and integration in other systems would be possible.

Three important use cases came to our mind:

- connecting SteVe with smart devices, making automation possible. For examples see [ifttt.com](https://ifttt.com/explore)
- integration of SteVe in a micro service architecture
- create a solid base for a possible frontend modernization, probably a Single Page Application

#### Additional context

This feature request was discussed between @goekay and @lextira in a video call, and it was agreed to cover two key features for now:
- create, read, update, delete of RFID-Tokens from a third party service (key feature for authorization automation)
- read transactions (key feature for reporting automation)

goekay, 2022-09-17:
ToDos:

- [ ] REST Controller for OCPP Tags with CRUD functionality (i.e. what OCPP Tags page under Data Management offers: same querying possibilities, same response data models). JSON payloads.
- [ ] REST Controller to read transaction info (i.e. what Transactions page under Data Management offers: same querying possibilities, same response data model). JSON payloads.
- [ ] Swagger documentation for the endpoints.
- [ ] Authentication for these endpoints via a static API key that will be specified in `main.properties`

anirudh-ramesh, 2022-09-21:
Hi @goekay, can you please suggest how to contribute to this feature request? I had made an OpenAPI specification for an HTTP REST API and began implementing it through an API controller, but that is WIP and nowhere near completion. However, I had completed adding Swagger UI to SteVe. I would be glad to contribute a few snippets if that is suitable for you.

anirudh-ramesh, 2022-09-21:
The endpoints that had been implemented and manually tested are as follows:
- Read charger properties (filter for active)
- Read tag properties
- Read transaction properties (filter for ongoing transactions)
- Start a transaction i.e. create a task to start a transaction
- Read last intermediate energy meter value and electricity consumed (kWh or Wh depending on the charger)
- Stop a transaction i.e. create a task to stop a transaction

goekay, 2022-09-29:
hey @anirudh-ramesh, thanks for your work! this feature request and its realisation has a very clear scope, and therefore i would like to keep it that way. while working on this, i am also laying some ground work for further extensions (such as auth and openapi spec).

if you'd like, you are welcome to make a PR (after this is finished and merged) of your impl where your endpoints use the ground work introduced by this one. thanks!

frmunozForcast, 2022-10-06:
Hi @goekay, great work here, I'm very interested in this feature request. How long do you think it will take to complete it for an usable scenario?. Thanks.

goekay, 2022-10-06:
@frmunozForcast it is usable already ;)

iprastha, 2022-10-19:
Hi, is this feature already released or under a specific branch? Thank you!

goekay, 2022-10-19:
hey @iprastha there is no release yet, but you can see the referenced commits above in the respective feature branch.

cbrandlehner, 2022-11-23:
@goekay am I right, that the next step is to create a pull request so that the commits can be merged and released?

goekay, 2022-11-23:
yes, that is correct. i am waiting for the testing by and feedback from @kohlerdominik to make a PR and merge it. as far as can see, the feature impl is complete.

kohlerdominik, 2022-11-23:
Hi all

Sorry for my delay. Just this week I started testing and implementation on our end. I think there are some minor things I want to give as feedback to @goekay, so he can consider wether he want's to adjust them, and then we're good to go. I will finish feedback until the end of the week.

However, by simply checking out the branch [910-api-endpoints-for-automation-and-integration](https://github.com/steve-community/steve/tree/910-api-endpoints-for-automation-and-integration), everyone can already use it.

cbrandlehner, 2022-11-24:
@kohlerdominik I tried to but found no documentation on the new APIs. All I found was a configuration option for the API key and token.

goekay, 2022-11-24:
true, that's an area for improvement. i try to write useful commit messages though. you can see that, if you checkout the branch and start steve, the endpoint `/manager/v3/api-docs` will show you the API spec.

goekay, 2022-12-08:
Hey everyone, this feature is merged and part of master branch. 

Thanks to @enertech-ch for sponsoring the feature, and to @kohlerdominik for the collaboration!

juherr, 2022-12-10:
This feature is awesome and open a lot of new options. Well done 👍

goekay, 2022-12-10:
> @kohlerdominik I tried to but found no documentation on the new APIs. All I found was a configuration option for the API key and token.

@cbrandlehner i added [a section](https://github.com/steve-community/steve/wiki/Configuration#rest-apis) about this. is it good enough for you to get started? the majority of the integration work should come from inspecting open-api spec and its capabilities. the documentation is the spec.