steve-community/steve #916: Transaction losing WebSocket connection when using any chargeport ID besides "2"

Khavatari93, 2022-09-19:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Ubuntu Server 22.04
JDK               : OpenJDK 11.0.16
Database          : MariaDB
```

#### Expected Behavior
When scanning a tag at the charging station it should start charging and Steve should show an active transaction
...

#### Actual Behavior
After scanning the tag, the charger waits for SteVe approval but never gets it hanging on authentication.
SteVe meanwhile lost the WebSocket connection because of:
"java.lang.IllegalArgumentException: 1 is an unknown message type"
...

#### Steps to Reproduce the Problem

  1. Use any charging port uid besides "2" to start transaction
  2. If you use connector ID 2 it starts the transaction (tested with charger emulator)

#### Additional context

Full Error from Logs:

```
[INFO ] 2022-09-19 12:13:54,161 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=EVB-P20261797, sessionId=55d64502-f988-fc87-cfab-4ad234573604] Received: [1,"oDvyPKkxnQSngCgmMDNsWOABfmughshvnR2j","StartTransaction",{"connectorId":1,"idTag":"1234","timestamp":"2022-09-19T14:13:52Z","meterStart":0,"reservationId":0}]
[ERROR] 2022-09-19 12:13:54,162 org.springframework.web.socket.adapter.jetty.Jetty10WebSocketHandlerAdapter - Closing session due to exception for JettyWebSocketSession[id=55d64502-f988-fc87-cfab-4ad234573604, uri=ws://172.16.40.220:8080/steve/websocket/CentralSystemService/EVB-P20261797]
java.lang.IllegalArgumentException: 1 is an unknown message type
	at de.rwth.idsg.steve.ocpp.ws.data.MessageType.fromTypeNr(MessageType.java:43) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.accept(Deserializer.java:75) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:49) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:114) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.onMessage(AbstractWebSocketEndpoint.java:86) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ConcurrentWebSocketHandler.handleMessage(ConcurrentWebSocketHandler.java:50) ~[steve.jar:?]
	at org.springframework.web.socket.adapter.jetty.Jetty10WebSocketHandlerAdapter.onWebSocketText(Jetty10WebSocketHandlerAdapter.java:78) ~[spring-websocket-5.3.18.jar:5.3.18]
	at org.eclipse.jetty.websocket.core.internal.messages.StringMessageSink.accept(StringMessageSink.java:53) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.acceptMessage(JettyWebSocketFrameHandler.java:334) ~[websocket-jetty-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.onTextFrame(JettyWebSocketFrameHandler.java:407) ~[websocket-jetty-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.onFrame(JettyWebSocketFrameHandler.java:240) ~[websocket-jetty-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession$IncomingAdaptor.lambda$onFrame$0(WebSocketCoreSession.java:660) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.server.handler.ContextHandler.handle(ContextHandler.java:1463) ~[jetty-server-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.server.handler.ContextHandler.handle(ContextHandler.java:1482) ~[jetty-server-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.server.internal.AbstractHandshaker$1.handle(AbstractHandshaker.java:212) ~[websocket-core-server-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession$IncomingAdaptor.onFrame(WebSocketCoreSession.java:660) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.AbstractExtension.nextIncomingFrame(AbstractExtension.java:144) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.PerMessageDeflateExtension.nextIncomingFrame(PerMessageDeflateExtension.java:225) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.PerMessageDeflateExtension$IncomingFlusher.inflate(PerMessageDeflateExtension.java:451) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.PerMessageDeflateExtension$IncomingFlusher.transform(PerMessageDeflateExtension.java:395) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.TransformingFlusher$Flusher.process(TransformingFlusher.java:127) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.IteratingCallback.processing(IteratingCallback.java:232) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.IteratingCallback.iterate(IteratingCallback.java:214) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.TransformingFlusher.sendFrame(TransformingFlusher.java:75) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.PerMessageDeflateExtension.onFrame(PerMessageDeflateExtension.java:91) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.ExtensionStack.onFrame(ExtensionStack.java:101) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession.onFrame(WebSocketCoreSession.java:476) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFrame(WebSocketConnection.java:262) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.fillAndParse(WebSocketConnection.java:458) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFillable(WebSocketConnection.java:343) ~[websocket-core-common-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:319) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:100) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.io.SocketChannelEndPoint$1.run(SocketChannelEndPoint.java:101) ~[jetty-io-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.runTask(AdaptiveExecutionStrategy.java:412) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.consumeTask(AdaptiveExecutionStrategy.java:381) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.tryProduce(AdaptiveExecutionStrategy.java:268) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.lambda$new$0(AdaptiveExecutionStrategy.java:138) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:407) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:894) ~[jetty-util-10.0.7.jar:10.0.7]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1038) ~[jetty-util-10.0.7.jar:10.0.7]
	at java.lang.Thread.run(Thread.java:829) ~[?:?]
[WARN ] 2022-09-19 12:13:54,163 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=EVB-P20261797, sessionId=55d64502-f988-fc87-cfab-4ad234573604] Connection is closed, status: CloseStatus[code=1011, reason=null]
[INFO ] 2022-09-19 12:13:54,175 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=EVB-P20261797, sessionId=7633fc8d-082f-35cc-2063-7e219237be57] Connection is established
```

goekay, 2022-09-22:
this issue is not about the id of the charging station or the connector. it's a technical problem.

the ocpp json spec describes a message structure. there is this notion of message type, and it is represented by the first element in the json array (which is the message payload).

1 is not a valid message type. that's the problem. where is this even coming from? it cannot imagine this to be happening with a proper charging station.

[image]

goekay, 2022-10-01:
i consider this issue resolved, and am closing it.

closed 2022-10-01 by goekay