steve-community/steve #918: Error - MySQL and Java are not using the same time zone

gklank, 2022-09-20:
#### Checklist

- [x ] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ ?] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Ubuntu 22.04 on raspberry pi 4
JDK               : openjdk 11.0.16 2022-07-19
Database          : MariaDB 10.6
```

#### Expected Behavior

...

#### Actual Behavior

raspberrypi-1:~/steve-4.6.9/steve-steve-3.4.9# java -jar target/steve.jar
Log file: /root/logs/steve.log
Starting................................................ FAILED!
Please refer to the log file for details

#### Steps to Reproduce the Problem

  1. tried several options in the mariadb config, still not working
  2. tried several opions for java and mvn, still not working

#### Additional context

any help would be appreciated...

shadsnz, 2022-09-20:
I'd suggest checking that the timezone for MySQL and the SteVe database are both set to UTC.   Also check the log file and post the relevant section showing the specific error(s).

Quote from the readme:

Important: Make sure that the time zone of the MySQL server is the same as [the time zone of SteVe].Since UTC is strongly recommended by OCPP, it is the default in SteVe and you should set it in MySQL, accordingly.

gklank, 2022-09-20:
hi,

many thanks!

A piece struggling with the timezones, but yes, changed to UTC and Steve is running fine!!!

Thanks and regars,

Gerhard

closed 2022-09-20 by gklank